awesometic/realtek-r8125-dkms #3: Driver seems to conflict with RPI cm4 onboard wifi

steeley, 2021-03-28:
running on raspberry pi buster with RPI Compute module 4 and CM4 I/o board.

card seems to work but  causes issues with the onboard cm4 wifi.

I'm using command line only buster with networkManager nmcli for setting up wifi. Works very well without the
8125 card in the PCIE slot. When the 8125 card is in the slot It stops me connecting to my wifi as a client or putting my cm4
wireless in access point mode. Again my wifi works perfectly without the 8125 card. but .... the 8125 seems to work ok
on its own if I forget about wifi and use it as a wired connection.

Very very odd.... Any clues what to look for?
thanks.

awesometic, 2021-03-29:
Hi.

Unfortunately, I have no clue either. And I don't have Raspberry Pi 4 CM. :sweat_smile: 
But I think there's a question we have to know clearly to narrow down the root cause: Is it a hardware issue or a software issue?

I suggest,
If it is in hardware, seems another PCIe device should occur the same error too. Then the root cause must be kind of a circuit problem.
If it is in software, then seems you should try another Linux distro for Raspberry Pi 4. And I think that should have a 5.10 kernel at least because the modern Linux kernel has the firmware binary for r8125 2.5GbE ethernet card. So that we can filter out one suspicious thing from the potential culprits.

Hope this helps.

steeley, 2021-03-29:
Thanks for the suggestions. I have another system so I will try it with that and see if I get the same issue. I can put a fresh install of Linux/Raspbian on that and see what happens.

closed 2021-05-25 by awesometic