awesometic/realtek-r8125-dkms #5: 9.005.05更新了

LsnmxNB, 2021-05-10:
https://www.realtek.com/zh-tw/component/zoo/category/network-interface-controllers-10-100-1000m-gigabit-ethernet-pci-express-software

awesometic, 2021-05-13:
Hi. Thanks for notifying me.

Actually I have seen that "ring driver" is releasing with a higher version than the Linux driver.
But I'm not sure what it is. I googled but no proper answer found.

Do you know what the Realtek ring driver is?

LsnmxNB, 2021-05-13:
> Hi. Thanks for notifying me.
> 
> Actually I have seen that "ring driver" is releasing with a higher version than the Linux driver.
> But I'm not sure what it is. I googled but no proper answer found.
> 
> Do you know what the Realtek ring driver is?

Hello, Others shows 9.005.05

[image]

LsnmxNB, 2021-05-13:
> > > Hi. Thanks for notifying me.
> > > Actually I have seen that "ring driver" is releasing with a higher version than the Linux driver.
> > > But I'm not sure what it is. I googled but no proper answer found.
> > > Do you know what the Realtek ring driver is?
> > 
> > 
> > Hello, Others shows 9.005.05
> > [image]
> 
> 作者其实只是想知道 Ring Library driver 和 Linux driver 有什么不一样，其实我也有相同的疑问。

我也想知道啊，群上一帮人说更新了

OiCkilL, 2021-05-21:
> 我也想知道啊，群上一帮人说更新了

刚发邮件问了下Realtek，Ring library driver 是针对高通cpu某些特殊场景下的应用，一般用上面的 Linux 版就行。

@awesometic Just sent an email to ask the Realtek supports. Ring library driver is for Qualcomm cpu some special scenarios of the application. Not for general use scenarios.

awesometic, 2021-05-21:
@OiCkilL Well noted. Thank you for asking Realtek, now we can know what Ring Library is clearly. 😄 

The name and the version number are so ambiguous. Looks like we can use that as a generic Linux driver but it is not.

closed 2021-05-25 by awesometic