awesometic/realtek-r8125-dkms #6: [Info] Building current version works with 20.04.1 and hwe kernel 5.8.0-55

pannal, 2021-06-08:
Just saw the note in your current release:
> I stopped Launchpad daily building for now because this version cannot build on the latest system, at least Ubuntu 21.04 + 5.11 kernel that I tested with. I will investigate this one to work with the current system. For now, to use this version of r8125 driver you should download/install it by yourself. In the case of Ubuntu 20.04 + 5.4 kernel, which is my other system, it can be updated flawlessly.

And wanted to confirm that it builds well using your private launchpad repo and the 5.8 kernel on 20.04.01 (using the hardware enablement stack, linux-generic-hwe-20.04).

awesometic, 2021-06-10:
Hi. Thanks for sharing your experience. :)
Good to know that Ubuntu 20.04 + 5.8 kernel is also not affected by the build problem.

closed 2021-06-10 by pannal

logopk, 2021-07-06:
I have read that news entries, that I found interesting:

"Ab Ubuntu 21.04 wird der RTL8125 nativ vom Kernel 5.10.x unterstützt" -> Ubuntu 21.04 r8125 native in kernel 5.10.x 
RealTek says it is only  "LINUX driver r8125 for kernel up to 5.6"

What do you think?

awesometic, 2021-07-22:
> I have read that news entries, that I found interesting:
> 
> "Ab Ubuntu 21.04 wird der RTL8125 nativ vom Kernel 5.10.x unterstützt" -> Ubuntu 21.04 r8125 native in kernel 5.10.x
> RealTek says it is only "LINUX driver r8125 for kernel up to 5.6"
> 
> What do you think?

Yeah, you can use RTL8125 when you are using kernel 5.10 or above out of the box. But the RTL8125 driver in the modern kernel is not a driver itself, rather it is provided as a firmware file.

So I think we can expect a more solid work of the RTL8125 module by using the dedicated driver from Realtek. And we also can get the latest features, improvements of the driver while the Linux firmware files have to be updated by Realtek to the Linux firmware maintainers.