awesometic/realtek-r8125-dkms #7: Wrong driver loaded (r8169 instead of r8125)

pannal, 2021-06-09:
Hey,

I've read that you've removed the explicit removal of the r8169 driver from the original autorun.sh.

This seems to conflict on my system (20.04.1 LTS, HWE kernel); when both modules are present, my Trendnet R8125 is loaded with the r8169 driver instead of the correct one.

I've now removed it manually, is this intentional behaviour?

awesometic, 2021-06-10:
Actually, that isn't intended. I thought that the system should load both two drivers at the same time but sounds it is not.
Sorry for the inconvenience. I will edit `autorun.sh` back to the original state by your actual experience. 

For those who read this issue thread: This is not related to the Debian packaging way. :)

closed 2021-06-14 by awesometic

awesometic, 2021-06-14:
I found that it also affects delivering the driver as a DKMS package. If the r8169 driver exists on the system then the r8125 driver will not load properly. I should update my Debian packaging scripts to add some codes to remove the existing legacy driver. Anyway, thank you for the report. :smile:

terr72, 2021-07-15:
@awesometic I need to run r8169 and r8125 simultaneously, since I have two NICs in my computer: one integrated RTL8168 and an addon-card iwth a RTL8125B chipset. Is this not possible?

awesometic, 2021-09-29:
@terr72 ,

Sorry for the late reply. How about removing the RTL8125 firmware files from the `/lib/firmware/rtl_nic` directory? Like the `sudo rm -f /lib/firmware/rtl_nic/rtl8125*` command. Then I think the `r8169` module won't load the firmware for the RTL8125 chipset so that it could make the `r8125` module working properly.

Hope this works. 🤔

shing-lee, 2023-09-19:
@awesometic 

`sudo echo 'alias r8169 off' >> /etc/modprobe.d/dkms.conf`
`sudo echo 'r8125' >> /etc/initramfs-tools/modules`
`sudo update-initramfs -u`

shing-lee, 2024-03-08:
> @awesometic我需要同时运行 r8169 和 r8125，因为我的计算机中有两个 NIC：一个集成 RTL8168 和一个带有 RTL8125B 芯片组的附加卡。这不可能吗？

sudo echo 'alias r8169 off' >> /etc/modprobe.d/dkms.conf
sudo echo 'r8125' >> /etc/initramfs-tools/modules
sudo update-initramfs -k all -u
sudo reboot

commit 32e1da19 (2021-06-14): Make autorun.sh to the original state and it fixes #7