awesometic/realtek-r8125-dkms #9: `error: 'struct rtl8125_private' has no member named 'lock'` when using `ENABLE_PTP_SUPPORT = y`

eirik-ff, 2021-09-24:
I need PTP for RTL8125 and tried to compile with `ENABLE_PTP_SUPPORT = y`, but get the following output and errors from `make`:

```
CC:  cc
CCVERSION:  9
KERNEL_GCC_VERSION: 
KVER:  5.11.0-36-generic
KMAJ:  5
KMIN:  11
KREV:  0
BASEDIR:  /lib/modules/5.11.0-36-generic
DRIVERDIR:  /lib/modules/5.11.0-36-generic/kernel/drivers/net/ethernet/realtek
PWD:  /home/[username]/realtek-r8125-dkms/src
RTKDIR:  kernel/drivers/net/ethernet/realtek

make -C /lib/modules/5.11.0-36-generic/build M=/home/[username]/realtek-r8125-dkms/src clean
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-36-generic'
  CLEAN   /home/[username]/realtek-r8125-dkms/src/Module.symvers
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-36-generic'
make -C /lib/modules/5.11.0-36-generic/build M=/home/[username]/realtek-r8125-dkms/src modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-36-generic'
  CC [M]  /home/[username]/realtek-r8125-dkms/src/r8125_n.o
  CC [M]  /home/[username]/realtek-r8125-dkms/src/rtl_eeprom.o
  CC [M]  /home/[username]/realtek-r8125-dkms/src/rtltool.o
  CC [M]  /home/[username]/realtek-r8125-dkms/src/r8125_ptp.o
In file included from ./include/linux/mmzone.h:8,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/umh.h:4,
                 from ./include/linux/kmod.h:9,
                 from ./include/linux/module.h:16,
                 from /home/[username]/realtek-r8125-dkms/src/r8125_ptp.c:35:
/home/[username]/realtek-r8125-dkms/src/r8125_ptp.c: In function ‘rtl8125_phc_adjtime’:
/home/[username]/realtek-r8125-dkms/src/r8125_ptp.c:133:30: error: ‘struct rtl8125_private’ has no member named ‘lock’
  133 |         spin_lock_irqsave(&tp->lock, flags);
      |                              ^~
./include/linux/spinlock.h:252:34: note: in definition of macro ‘raw_spin_lock_irqsave’
  252 |   flags = _raw_spin_lock_irqsave(lock); \
      |                                  ^~~~
/home/[username]/realtek-r8125-dkms/src/r8125_ptp.c:133:9: note: in expansion of macro ‘spin_lock_irqsave’
  133 |         spin_lock_irqsave(&tp->lock, flags);
      |         ^~~~~~~~~~~~~~~~~
/home/[username]/realtek-r8125-dkms/src/r8125_ptp.c:136:35: error: ‘struct rtl8125_private’ has no member named ‘lock’
  136 |         spin_unlock_irqrestore(&tp->lock, flags);
      |                                   ^~
In file included from ./include/linux/mmzone.h:8,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/umh.h:4,
                 from ./include/linux/kmod.h:9,
                 from ./include/linux/module.h:16,
                 from /home/[username]/realtek-r8125-dkms/src/r8125_ptp.c:35:
/home/[username]/realtek-r8125-dkms/src/r8125_ptp.c: In function ‘rtl8125_phc_gettime’:
/home/[username]/realtek-r8125-dkms/src/r8125_ptp.c:206:30: error: ‘struct rtl8125_private’ has no member named ‘lock’
  206 |         spin_lock_irqsave(&tp->lock, flags);
      |                              ^~
./include/linux/spinlock.h:252:34: note: in definition of macro ‘raw_spin_lock_irqsave’
  252 |   flags = _raw_spin_lock_irqsave(lock); \
      |                                  ^~~~
/home/[username]/realtek-r8125-dkms/src/r8125_ptp.c:206:9: note: in expansion of macro ‘spin_lock_irqsave’
  206 |         spin_lock_irqsave(&tp->lock, flags);
      |         ^~~~~~~~~~~~~~~~~
/home/[username]/realtek-r8125-dkms/src/r8125_ptp.c:208:35: error: ‘struct rtl8125_private’ has no member named ‘lock’
  208 |         spin_unlock_irqrestore(&tp->lock, flags);
      |                                   ^~
In file included from ./include/linux/mmzone.h:8,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/umh.h:4,
                 from ./include/linux/kmod.h:9,
                 from ./include/linux/module.h:16,
                 from /home/[username]/realtek-r8125-dkms/src/r8125_ptp.c:35:
/home/[username]/realtek-r8125-dkms/src/r8125_ptp.c: In function ‘rtl8125_phc_settime’:
[... 71 lines cut ...]
                 from ./include/linux/umh.h:4,
                 from ./include/linux/kmod.h:9,
                 from ./include/linux/module.h:16,
                 from /home/[username]/realtek-r8125-dkms/src/r8125_ptp.c:35:
/home/[username]/realtek-r8125-dkms/src/r8125_ptp.c:570:38: error: ‘struct rtl8125_private’ has no member named ‘lock’
  570 |                 spin_lock_irqsave(&tp->lock, flags);
      |                                      ^~
./include/linux/spinlock.h:252:34: note: in definition of macro ‘raw_spin_lock_irqsave’
  252 |   flags = _raw_spin_lock_irqsave(lock); \
      |                                  ^~~~
/home/[username]/realtek-r8125-dkms/src/r8125_ptp.c:570:17: note: in expansion of macro ‘spin_lock_irqsave’
  570 |                 spin_lock_irqsave(&tp->lock, flags);
      |                 ^~~~~~~~~~~~~~~~~
/home/[username]/realtek-r8125-dkms/src/r8125_ptp.c:572:43: error: ‘struct rtl8125_private’ has no member named ‘lock’
  572 |                 spin_unlock_irqrestore(&tp->lock, flags);
      |                                           ^~
make[2]: *** [scripts/Makefile.build:288: /home/[username]/realtek-r8125-dkms/src/r8125_ptp.o] Error 1
make[1]: *** [Makefile:1848: /home/[username]/realtek-r8125-dkms/src] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-36-generic'
make: *** [Makefile:176: modules] Error 2
```

Is there something I'm missing, or is PTP not supported yet by the driver? 

Thanks!

eirik-ff, 2021-09-28:
The `lock` member seems to be removed in the latest commit on `r8125.h` for some reason ([here](https://github.com/awesometic/realtek-r8125-dkms/commit/3fbdcb61fb588061a220b94c9af63e78d36be728#diff-1db9b97ab467a8b824e8e31f29dc1f69c61ff4be378adc36ce3d81d2fd3a0e62L1791) is the diff). Why was it removed?

awesometic, 2021-09-29:
Hello.

Thanks for the report. I had a look at that `tp->lock` and yeah, maybe Realtek forgot to remove the spinlock parts in the `r8125_ptp.c` file? I don't know the exact reason. 😄 

Since they simply removed all the spinlock lines in this version, you also may get it working by removing the same spinlock codes in the `r8125_ptp.c` file.
Or, we could add the `lock` member back to use PTP in a workaround. Actually, it is a slightly better way because we don't know the side effects of removing the spinlock on the PTP side.. Probably.

I can build a driver that is PTP enabled with one of those above methods. I will upload the patched file.

```
awesometic@awesometic-nas:Developer/realtek-r8125-dkms ‹master*›$ make -j 4 ENABLE_PTP_SUPPORT=y
/bin/sh: 1: VER: not found
make -C src/ KVER= BASEDIR=/lib/modules/5.11.0-34-generic clean
make -C src/ KVER= BASEDIR=/lib/modules/5.11.0-34-generic modules
make[1]: Entering directory '/home/awesometic/Developer/realtek-r8125-dkms/src'
make -C src/ KVER= BASEDIR=/lib/modules/5.11.0-34-generic install
make[1]: Entering directory '/home/awesometic/Developer/realtek-r8125-dkms/src'
make[1]: Entering directory '/home/awesometic/Developer/realtek-r8125-dkms/src'
make -C /lib/modules/5.11.0-34-generic/build M=/home/awesometic/Developer/realtek-r8125-dkms/src INSTALL_MOD_DIR=kernel/drivers/net/ethernet/realtek modules_install
make -C /lib/modules/5.11.0-34-generic/build M=/home/awesometic/Developer/realtek-r8125-dkms/src clean
make -C /lib/modules/5.11.0-34-generic/build M=/home/awesometic/Developer/realtek-r8125-dkms/src modules
make[2]: Entering directory '/usr/src/linux-headers-5.11.0-34-generic'
make[2]: Entering directory '/usr/src/linux-headers-5.11.0-34-generic'
make[2]: Entering directory '/usr/src/linux-headers-5.11.0-34-generic'
make[2]: Leaving directory '/usr/src/linux-headers-5.11.0-34-generic'
make[1]: Leaving directory '/home/awesometic/Developer/realtek-r8125-dkms/src'
cat: /home/awesometic/Developer/realtek-r8125-dkms/src/modules.order: No such file or directory
  DEPMOD  5.11.0-34-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
  CC [M]  /home/awesometic/Developer/realtek-r8125-dkms/src/r8125_n.o
make[2]: Leaving directory '/usr/src/linux-headers-5.11.0-34-generic'
make[1]: Leaving directory '/home/awesometic/Developer/realtek-r8125-dkms/src'
  CC [M]  /home/awesometic/Developer/realtek-r8125-dkms/src/rtl_eeprom.o
  CC [M]  /home/awesometic/Developer/realtek-r8125-dkms/src/rtltool.o
  CC [M]  /home/awesometic/Developer/realtek-r8125-dkms/src/r8125_ptp.o
  LD [M]  /home/awesometic/Developer/realtek-r8125-dkms/src/r8125.o
  MODPOST /home/awesometic/Developer/realtek-r8125-dkms/src/Module.symvers
  CC [M]  /home/awesometic/Developer/realtek-r8125-dkms/src/r8125.mod.o
  LD [M]  /home/awesometic/Developer/realtek-r8125-dkms/src/r8125.ko
make[2]: Leaving directory '/usr/src/linux-headers-5.11.0-34-generic'
make[1]: Leaving directory '/home/awesometic/Developer/realtek-r8125-dkms/src'
```

```
awesometic@awesometic-nas:realtek-r8125-dkms/src ‹master*›$ modinfo ./r8125.ko
filename:       /home/awesometic/Developer/realtek-r8125-dkms/src/./r8125.ko
version:        9.006.04-NAPI-PTP
license:        GPL
description:    Realtek RTL8125 2.5Gigabit Ethernet driver
author:         Realtek and the Linux r8125 crew <netdev@vger.kernel.org>
srcversion:     F20AD9EEFB9E43B7841BED8
alias:          pci:v000010ECd00003000sv*sd*bc*sc*i*
alias:          pci:v000010ECd00008125sv*sd*bc*sc*i*
depends:
retpoline:      Y
name:           r8125
vermagic:       5.11.0-34-generic SMP mod_unload modversions
parm:           speed_mode:force phy operation. Deprecated by ethtool (8). (uint)
parm:           duplex_mode:force phy operation. Deprecated by ethtool (8). (uint)
parm:           autoneg_mode:force phy operation. Deprecated by ethtool (8). (uint)
parm:           advertising_mode:force phy operation. Deprecated by ethtool (8). (uint)
parm:           aspm:Enable ASPM. (int)
parm:           s5wol:Enable Shutdown Wake On Lan. (int)
parm:           s5_keep_curr_mac:Enable Shutdown Keep Current MAC Address. (int)
parm:           rx_copybreak:Copy breakpoint for copy-only-tiny-frames (int)
parm:           use_dac:Enable PCI DAC. Unsafe on 32 bit PCI slot. (int)
parm:           timer_count:Timer Interrupt Interval. (int)
parm:           eee_enable:Enable Energy Efficient Ethernet. (int)
parm:           hwoptimize:Enable HW optimization function. (ulong)
parm:           s0_magic_packet:Enable S0 Magic Packet. (int)
parm:           tx_no_close_enable:Enable TX No Close. (int)
parm:           enable_ptp_master_mode:Enable PTP Master Mode. (int)
parm:           disable_pm_support:Disable PM support. (int)
parm:           debug:Debug verbosity level (0=none, ..., 16=all) (int)
```

I will upload the patched file soon. Please test that version then.

closed 2021-09-29 by awesometic

eirik-ff, 2021-09-29:
Seems like everything works fine with the patch. I can compile and `ethtool` reports hardware timestamping capabilities. 

One issue I have now is that the results for PTP is terrible on one PC using the RTL8125 chipset, while on another PC using the same chipset it works fine. But I guess that's not a driver issue, but rather hardware.

commit 62cbef03 (2021-09-29): Add spinlock_t lock member back to the r8125_private struct

Without this member it fails to build with the PTP feature.

This patch fixes #9.