awesometic/realtek-r8125-dkms #13: The package installation in Proxmox

lk7777, 2021-11-28:
I am getting the following message:

`Module build for kernel 5.13.19-1-pve was skipped since the kernel headers for this kernel does not seem to be installed.`

I have checked with the following command:

`apt search pve-headers-$(uname -r)`

It says that headers are installed.

`ls -l /usr/src`
linux-headers-5.13.19-1-pve

lk7777, 2021-11-28:
It works now. I forgot (at least, I think so) to run update-initramfs -u -k $(uname -r) after proxmox kernel headers installation. Thank you for this great package.

awesometic, 2021-11-29:
Great. Glad to hear you solve that.
I think maybe `dkms autoinstall` also would fix that problem. But don't know, there're so many possibilities in the Linux world :smile: 
I will close this issue.

closed 2021-11-29 by awesometic