awesometic/realtek-r8125-dkms #15: Incorrect make command does extra install

kalj, 2021-12-31:
I believe this line is not quite right:
https://github.com/awesometic/realtek-r8125-dkms/blob/master/dkms.conf#L5
By having the make target `all` there, the `install` target will be built as well resulting in the `.ko` file being copied a second time, to something like `/usr/lib/modules/5.XX.X-XX-generic/kernel/drivers/net/ethernet/realtek/r8125.ko`. For me this lead to a duplicate detection, resulting in the actual module not being installed, and some secure boot signature issues. I think the correct make target is `modules`, see here for instance:
https://aur.archlinux.org/cgit/aur.git/tree/dkms.conf?h=r8125-dkms
Changing that fixed it for me.

awesometic, 2022-01-02:
Hello. Thanks for the information. Yeah, seems it should be patched.
After testing in some cases I will patch that part soon.

awesometic, 2022-01-20:
I made a patch for correcting the mentioned make target.

Thanks again 😄

closed 2022-01-20 by awesometic