awesometic/realtek-r8125-dkms #21: Problem with ubuntu

bizoally, 2022-03-20:
I am using intel b660 ds3h motherboard and realtek r8125 but i am not able to use wifi with ubuntu do u have any solution for it ?

awesometic, 2022-03-21:
Sorry I don't get it. Are you going to use wifi? This r8125 module is for a wired Ethernet connection.

awesometic, 2022-03-22:
r8125 is not a wireless one. Are you sure you are using RTL8125 Ethernet?
What is the result of `lspci | grep -i ethernet` command?

bizoally, 2022-03-27:
PFAIt is showing realtek semiconductor co., Ltd. RTL8125 2.5GbE controller (rev 05)

awesometic, 2022-03-28:
What is PFAlt?
Sorry that I can't understand what you're saying 😅 

That is the proper identified name of our RTL8125 device, but RTL8125 is not a wireless one.
If you want to use wireless ethernet and if you are finding a proper driver for your Linux system, I'm sorry that this is not a proper place.

I think you may misread your hardware information.

I will close this issue for now,
If you have a problem with using this DKMS package, please make an issue again.

Thanks.