awesometic/realtek-r8125-dkms #23: NIC keeps turning off

wouterdebruijn, 2022-05-06:
System:
 Linux 5.10.0-13-amd64 SMP Debian 5.10.106-1 (2022-03-17) x86_64 GNU/Linux
 Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller
  Subsystem: ASRock Incorporation RTL8125 2.5GbE Controller

Behavior:
The NIC runs at 1000mb/s (limited by the connected device) and turns off after a couple of hours.

Expected behavior:
The NIC runs at 1000mb/s (limited by the connected device) and stays connected.

With the default system driver the nic runs at 100mb/s

awesometic, 2022-05-10:
Hello,

Actually, the purpose of this package is just for providing Realtek r8125 driver as an installable DKMS Debian package.
So, I'm sorry that I can't help you with a driver specific issue.

By the way, I think you can track your problem by checking `dmesg` on your Linux system.
I think if your router supports 2.5 GbE, and your Ethernet cable also does, then you should see the 2.5 Gbits working with your system.

wouterdebruijn, 2022-05-10:
Jup, sorry.

I confirmed that the driver downloaded from the official website had the same issue. But I forgot to close this issue.

Thank you for your reply!

closed 2022-05-10 by wouterdebruijn