awesometic/realtek-r8125-dkms #25: 大佬下一个版本更新能否开启rx和tx

8890564, 2022-06-09:
大佬下一个版本更新能否开启rx和tx，谢谢！

awesometic, 2022-06-09:
Hello,

I translated using Google 😅  =>
Can you enable rx and tx in the next version update, thank you!

What's the enabling rx and tx?
Do you mean that a feature using rx/tx?

If you want a new feature from the driver, I'm sorry that this package is for the DKMS package of the Realtek r8125 Linux driver so I can't support it. 😃

8890564, 2022-06-09:
感谢您的回复。好像从9.007.01开始官方驱动默认关闭了RSS（rx 的多队列支持，）也关闭了TX的多队列，RTL8125B硬件上是支持32个队列的，Makefile参数ENABLE_MULTIPLE_TX_QUEUE = n和ENABLE_RSS_SUPPORT =n
实测开启后仍支持RSS。

awesometic, 2022-06-10:
Ah okay, I understand. Then, do you want a package that enabled `ENABLE_MULTIPLE_TX_QUEUE` and `ENABLE_RSS_SUPPORT`?

I don't know them exactly, is there any side effect with them?
Or, you can build the driver yourself after cloning my repository.

pikachullk, 2022-07-05:
> Ah okay, I understand. Then, do you want a package that enabled `ENABLE_MULTIPLE_TX_QUEUE` and `ENABLE_RSS_SUPPORT`?
> 
> I don't know them exactly, is there any side effect with them? Or, you can build the driver yourself after cloning my repository.

Actually he means that
change 
ENABLE_MULTIPLE_TX_QUEUE = n
ENABLE_RSS_SUPPORT = n
from n to y in file [realtek-r8125-dkms](https://github.com/awesometic/realtek-r8125-dkms)/[src](https://github.com/awesometic/realtek-r8125-dkms/tree/master/src)/Makefile
Because it‘s disable by default from 9.007.01
But yeah~just edit this makefile after cloning is okay as well~

awesometic, 2022-07-08:
Thanks, I didn't notice that is disabled since 9.007.01.

I can prepare another package besides the existing package. Then I think I should gather the features people would want, such as disabling ASPM.

Is there another feature that can be enabled in the new package? The name would be, `realtek-r8125-dkms-tune`, is it good in English?

pikachullk, 2022-07-08:
> Thanks, I didn't notice that is disabled since 9.007.01.
>
> I can prepare another package besides the existing package. Then I think I should gather the features people would want, such as disabling [...]

That sounds awesome bro!!!
Would you please notice us as soon as this "realtek-r8125-dkms-tune" release~
thank you for your driver again~

pikachullk, 2022-07-08:
> Thanks, I didn't notice that is disabled since 9.007.01.
>
> I can prepare another package besides the existing package. Then I think I should gather the features people would want, such as disabling [...]

What's more~I found that no matter if I set CONFIG_ASPM = y or n 
After I install the driver~
when I type~
modinfo r8125 
there still showing
parm:           aspm:Enable ASPM. (int)
Is that mean aspm is fail to disable?
So what should I do to disable it?

crazyshitass, 2022-07-10:
2.5G Ethernet LINUX driver r8125 for kernel up to 5.19 | 9.009.02

awesometic, 2022-07-10:
> What's more,I found that no matter if I set CONFIG_ASPM = y or n After I install the driver~ when I type~ modinfo r8125 there still showing parm: aspm:Enable ASPM. (int) Is that mean aspm is fail to [...]

It looks like that `parm` list in `modinfo` shows us the available module parameters, not the current value.
In the source codes, it lists all the parameters regardless of which parameter actually turns on or off now.
We can get the parameters using sysfs but there are no parameter files for r8125.

But by editing the configs in `src/Makefile`, it should take effect when the compile time.

awesometic, 2022-07-10:
> 2.5G Ethernet LINUX driver r8125 for kernel up to 5.19 | 9.009.02

Thanks for the information. I updated to the latest one. :)

crazyshitass, 2022-07-12:
[realtek-r8125-dkms_9.009.02-1_amd64.deb.zip](https://github.com/awesometic/realtek-r8125-dkms/files/9091005/realtek-r8125-dkms_9.009.02-1_amd64.deb.zip)
I personally build one with ENABLE_MULTIPLE_TX_QUEUE = y ,ENABLE_RSS_SUPPORT = y, CONFIG_ASPM = n. JUST UNZIP the file and install. Hope it works :)

pikachullk, 2022-07-12:
> [realtek-r8125-dkms_9.009.02-1_amd64.deb.zip](https://github.com/awesometic/realtek-r8125-dkms/files/9091005/realtek-r8125-dkms_9.009.02-1_amd64.deb.zip) I personally build one with [...]

I appreciate it ！！！My sweet crazyshitass!!!
This really help!!Thanks again & again!!!

LsnmxNB, 2022-09-20:
> > 2.5G Ethernet LINUX driver r8125 for kernel up to 5.19 | 9.009.02
> 
> Thanks for the information. I updated to the latest one. :)

In the next update, it is recommended to provide amd64.deb with RSS enabled, thank you

barkmann44, 2022-11-04:
hello!   9.010.01 has just been released.  Thankyou for your project!

LsnmxNB, 2022-11-09:
可以了，谢谢

[image]
[image]

smdx, 2022-12-15:
> 可以了，谢谢
> 
> [image] [image]

你好，请问这个realtek-r8125-dkms_9.010.01-1版本那里可以下载啊？
在releases里面下载的版本Makefile文档里面CONFIG_ASPM还是y其他参数也没有修改。

awesometic, 2022-12-15:
> 你好，请问这个realtek-r8125-dkms_9.010.01-1版本那里可以下载啊？ 在releases里面下载的版本Makefile文档里面CONFIG_ASPM还是y其他参数也没有修改。

There isn't yet.
I need to prepare the script for releasing both stock one and the tuned one simultaneously, with the same PPA repository.

Temporarily, I attached the tuned version here for you.
Edited 3 configs,
- `CONFIG_ASPM` to `n`
- `ENABLE_MULTIPLE_TX_QUEUE` to `y`
- `ENABLE_RSS_SUPPORT` to `y`

[realtek-r8125-dkms_9.010.01-1_amd64.deb.zip](https://github.com/awesometic/realtek-r8125-dkms/files/10234490/realtek-r8125-dkms_9.010.01-1_amd64.deb.zip)

Please tell me if it doesn't work as it should with the edited configs :)

smdx, 2022-12-15:
> There isn't yet. I need to prepare the script for releasing both stock one and the tuned one simultaneously, with the same PPA repository.
>
> Temporarily, I attached the tuned version here for you. [...]

The network service cannot be interrupted yet, I will install this driver later. thank you very much

smdx, 2022-12-27:
> 

Hello, the driver release version has reached 9.010.01-2. Can you take the time to upload an edited version of the configuration? thank you very much.

- CONFIG_ASPM to n
- ENABLE_MULTIPLE_TX_QUEUE to y
- ENABLE_RSS_SUPPORT to y

awesometic, 2022-12-28:
> > 
> 
> Hello, The driver release version has reached 9.010.01-2. Can you take the time to upload the edited version? thank you very much.
> 
>     * CONFIG_ASPM to n
> 
>     * ENABLE_MULTIPLE_TX_QUEUE to y
> 
>     * ENABLE_RSS_SUPPORT to y

Sure, here you are.

[realtek-r8125-dkms_9.010.01-2_amd64.zip](https://github.com/awesometic/realtek-r8125-dkms/files/10311437/realtek-r8125-dkms_9.010.01-2_amd64.zip)

smdx, 2022-12-28:
That's great, thank you very much.

licocofan, 2023-03-24:
realtek官方版本从9.007.01开始默认关闭了RSS，r8125_n.c以下代码片段确定了EnableRss为0，关闭了RSS功能。最终只能使用1个rx队列 2个tx队列了，HwSuppNumTxQueues还是2。

#ifndef ENABLE_LIB_SUPPORT
        switch (tp->mcfg) {
        case CFG_METHOD_4:
        case CFG_METHOD_5:
        case CFG_METHOD_7:
                if (tp->HwSuppIsrVer == 2) {
                        tp->RequireRduNonStopPatch = 1; 
                       tp->EnableRss = 0; 
               }
                break; 
       }
#endif

suhdd, 2023-05-05:
> What's more,I found that no matter if I set CONFIG_ASPM = y or n After I install the driver~ when I type~ modinfo r8125 there still showing parm: aspm:Enable ASPM. (int) Is that mean aspm is fail to [...]

I have the same problem on version 9.011.01 and 9.010.01
no matter i set the CONFIG_ASPM to 'y' or 'n'.  modinfo always show "parm:           aspm:Enable ASPM. (int)"
```
root@pve:/usr/src/r8125-9.011.01/src# modinfo r8215
modinfo: ERROR: Module r8215 not found.
root@pve:/usr/src/r8125-9.011.01/src# modinfo r8125
filename:       /lib/modules/5.15.104-1-pve/updates/dkms/r8125.ko
version:        9.011.01-NAPI-RSS
license:        GPL
description:    Realtek RTL8125 2.5Gigabit Ethernet driver
author:         Realtek and the Linux r8125 crew <netdev@vger.kernel.org>
srcversion:     E26366B3A512ACEBC7008DD
alias:          pci:v000010ECd00005000sv*sd*bc*sc*i*
alias:          pci:v000010ECd00008126sv*sd*bc*sc*i*
alias:          pci:v000010ECd00003000sv*sd*bc*sc*i*
alias:          pci:v000010ECd00008162sv*sd*bc*sc*i*
alias:          pci:v000010ECd00008125sv*sd*bc*sc*i*
depends:        
retpoline:      Y
name:           r8125
vermagic:       5.15.104-1-pve SMP mod_unload modversions 
parm:           speed_mode:force phy operation. Deprecated by ethtool (8). (uint)
parm:           duplex_mode:force phy operation. Deprecated by ethtool (8). (uint)
parm:           autoneg_mode:force phy operation. Deprecated by ethtool (8). (uint)
parm:           advertising_mode:force phy operation. Deprecated by ethtool (8). (uint)
parm:           aspm:Enable ASPM. (int)
parm:           s5wol:Enable Shutdown Wake On Lan. (int)
parm:           s5_keep_curr_mac:Enable Shutdown Keep Current MAC Address. (int)
parm:           use_dac:Enable PCI DAC. Unsafe on 32 bit PCI slot. (int)
parm:           timer_count:Timer Interrupt Interval. (int)
parm:           eee_enable:Enable Energy Efficient Ethernet. (int)
parm:           hwoptimize:Enable HW optimization function. (ulong)
parm:           s0_magic_packet:Enable S0 Magic Packet. (int)
parm:           tx_no_close_enable:Enable TX No Close. (int)
parm:           enable_ptp_master_mode:Enable PTP Master Mode. (int)
parm:           disable_pm_support:Disable PM support. (int)
parm:           enable_double_vlan:Enable Double VLAN. (int)
parm:           debug:Debug verbosity level (0=none, ..., 16=all) (int)

```

awesometic, 2023-05-08:
That message is from `MODULE_PARM_DESC` function, which is for describing the available parameters. It may not be the current state.

By the way, we couldn't know the currently activated parameters. It should be under the`/sys/module/r8125` directory but there is no `parameters` thing 🤔

eabase, 2024-12-21:
Any reason why the keep wake on LAN (WOL) disabled? 

```bash
parm:           s5wol:Enable Shutdown Wake On Lan. (int)
...
parm:           s0_magic_packet:Enable S0 Magic Packet. (int)
```