awesometic/realtek-r8125-dkms #26: Clarify installation instructions.

asokolsky, 2022-08-20:
I follow instructions from README:

On [releases page](https://github.com/awesometic/realtek-r8125-dkms/releases)
get the link to the latest deb package, e.g.
https://github.com/awesometic/realtek-r8125-dkms/releases/download/9.009.02-1/realtek-r8125-dkms_9.009.02-1_amd64.deb

```
root@pmox3:~# wget https://github.com/awesometic/realtek-r8125-dkms/releases/download/9.009.02-1/realtek-r8125-dkms_9.009.02-1_amd64.deb
```

Fix broken config:

```
root@pmox3:~/r8125-9.009.02# apt --fix-broken install
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

Install the driver:

```
root@pmox3:~# dpkg -i realtek-r8125-dkms_9.009.02-1_amd64.deb
(Reading database ... 89325 files and directories currently installed.)
Preparing to unpack realtek-r8125-dkms_9.009.02-1_amd64.deb ...

------------------------------
Deleting module version: 9.009.02
completely from the DKMS tree.
------------------------------
Done.
Unpacking realtek-r8125-dkms (9.009.02-1) over (9.009.02-1) ...
Setting up realtek-r8125-dkms (9.009.02-1) ...
Loading new realtek-r8125-9.009.02 DKMS files...
Building for 5.15.39-3-pve
Building for architecture amd64
Module build for kernel 5.15.39-3-pve was skipped since the
kernel headers for this kernel does not seem to be installed.
```

How do I gauge the success?

awesometic, 2022-09-13:
Hello,

As far as I know, you can reuse the built module in another machine for the security problem.
You should build it on each server, installing using the DKMS method will help you.

awesometic, 2022-11-07:
I will close this issue since there is no activity last about 2 months :)

closed 2022-11-07 by awesometic