awesometic/realtek-r8125-dkms #29: Proxmox v7.2 installation errors

Drauku, 2022-11-18:
I'm attempting to install these drivers on Proxmox v7.2-1 on Debian kernel 5.15.30-2. I've also tried it on kernel 5.15-64, which results in different errors about kernel headers not being present despite them obviously being in /usr/src/.
My first attempt at installing was to follow the recommended path of adding the awesometic repository, but I cannot get this to work because there is no valid OpenPGP data found:
```
root@proxmox:~# add-apt-repository ppa:awesometic/ppa
 Glad to see you.

With this PPA you can install the following packages.
- realtek-r8125-dkms
- realtek-r8152-dkms

If there's an issue during using my packages, please let me know using the Github repository.
- https://github.com/awesometic
You can see the repositories for each package.

Thank you :)
 More info: https://launchpad.net/~awesometic/+archive/ubuntu/ppa
Press [ENTER] to continue or ctrl-c to cancel adding it

gpg: keybox '/tmp/tmpcqodkpqd/pubring.gpg' created
gpg: /tmp/tmpcqodkpqd/trustdb.gpg: trustdb created
gpg: key B44E6C80FF7107B6: public key "Launchpad PPA for Yang Deokgyu" imported
gpg: Total number processed: 1
gpg:               imported: 1
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
gpg: no valid OpenPGP data found.
root@proxmox:~# apt install realtek-r8125-dkms
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package realtek-r8125-dkms
root@proxmox:~#
```

Current DKMS installation attempt using `./dkms-install.sh` results in this error:
```
Error!  Build of r8125.ko failed for: 5.15.30-2-pve (x86_64)
Make sure the name of the generated module is correct and at the root of the
build directory, or consult make.log in the build directory
/var/lib/dkms/r8125/9.010.01/build/ for more information.
```

In order to remove the error, I've tried installation manually by tar extraction, dkms.conf creation, and dkms add/build/install commands, but receive the same error.
I checked the `make.log` file in the indicated directory, but it was unhelpful:
```
DKMS make.log for r8125-9.010.01 for kernel 5.15.30-2-pve (x86_64)
Fri Nov 18 19:35:29 UTC 2022
make: Entering directory '/usr/src/linux-headers-5.15.30-2-pve'
  MODPOST /var/lib/dkms/r8125/9.010.01/build/Module.symvers
make: Leaving directory '/usr/src/linux-headers-5.15.30-2-pve'
```

I was able to successfully force the installation of the driver using the included `./install.sh` script from Realtek, but I understand that will not keep the driver working after kernel upgrades, and I would like to have the ability to upgrade my Proxmox host and maintain the connectivity of the OPNsense VM.
Do you have any suggestions for a solution?

awesometic, 2022-11-21:
Hi,

That is weird 🤔
Do you have the directory `/usr/src/realtek-r8125-9.010.01`?
If you have, how about `sudo dkms install -m realtek-r8125/9.010.01`?

awesometic, 2024-01-22:
I will close this issue since it was created about a year ago. If you have another problem with this package, please make a new issue.

closed 2024-01-22 by awesometic

Drauku, 2024-01-22:
My apologies, I completely missed your replay _two years ago__!

Because of the inability to upgrade my kernel, I have not updated my Proxmox install since the driver install.

Thank you for replying to me, and I will refer back to this once I rebuild my OPNsense virtual machine.