awesometic/realtek-r8125-dkms #31: DKMS doesn't compile on PVE-Kernel-6.1

stich86, 2022-12-18:
Hi,

i want to try opt-in 6.1 kernel on PVE 7.3, but the DKMS driver doesn't compile, here is the error:

```
DKMS make.log for realtek-r8125-9.010.01 for kernel 6.1.0-1-pve (x86_64)
Sun 18 Dec 2022 06:20:13 PM CET
/bin/sh: 1: VER: not found
make -C src/ KVER=6.1.0-1-pve BASEDIR=/lib/modules/6.1.0-1-pve modules
make[1]: Entering directory '/var/lib/dkms/realtek-r8125/9.010.01/build/src'
make -C /lib/modules/6.1.0-1-pve/build M=/var/lib/dkms/realtek-r8125/9.010.01/build/src modules
make[2]: Entering directory '/usr/src/linux-headers-6.1.0-1-pve'
  CC [M]  /var/lib/dkms/realtek-r8125/9.010.01/build/src/r8125_n.o
  CC [M]  /var/lib/dkms/realtek-r8125/9.010.01/build/src/rtl_eeprom.o
  CC [M]  /var/lib/dkms/realtek-r8125/9.010.01/build/src/rtltool.o
In file included from /var/lib/dkms/realtek-r8125/9.010.01/build/src/r8125_n.c:83:
/var/lib/dkms/realtek-r8125/9.010.01/build/src/r8125_n.c: In function ‘rtl8125_init_napi’:
/var/lib/dkms/realtek-r8125/9.010.01/build/src/r8125.h:636:57: error: too many arguments to function ‘netif_napi_add’
  636 | #define RTL_NAPI_CONFIG(ndev, priv, function, weight)   netif_napi_add(ndev, &priv->napi, function, weight)
      |                                                         ^~~~~~~~~~~~~~
/var/lib/dkms/realtek-r8125/9.010.01/build/src/r8125_n.c:12614:17: note: in expansion of macro ‘RTL_NAPI_CONFIG’
12614 |                 RTL_NAPI_CONFIG(tp->dev, r8125napi, poll, R8125_NAPI_WEIGHT);
      |                 ^~~~~~~~~~~~~~~
In file included from /var/lib/dkms/realtek-r8125/9.010.01/build/src/r8125_n.c:42:
./include/linux/netdevice.h:2562:1: note: declared here
 2562 | netif_napi_add(struct net_device *dev, struct napi_struct *napi,
      | ^~~~~~~~~~~~~~
make[3]: *** [scripts/Makefile.build:258: /var/lib/dkms/realtek-r8125/9.010.01/build/src/r8125_n.o] Error 1
make[2]: *** [Makefile:1996: /var/lib/dkms/realtek-r8125/9.010.01/build/src] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-6.1.0-1-pve'
make[1]: *** [Makefile:176: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/realtek-r8125/9.010.01/build/src'
make: *** [Makefile:42: modules] Error 2
```

Are you able to fix it or we need to wait Realtek to update the driver for 6.1?

Thx

awesometic, 2022-12-19:
It seems like it can be fixed quite simply.
I checked that `netif_napi_add` function in kernel 6.1, and I could see a slight change in the function parameter list.

I attached the patched version, could you check if it works?

[realtek-r8125-dkms_9.010.01-2_amd64.zip](https://github.com/awesometic/realtek-r8125-dkms/files/10255674/realtek-r8125-dkms_9.010.01-2_amd64.zip)

stich86, 2022-12-19:
hi @awesometic 

first.. thank you very much for your fast response.

I've tried again after changing the dpkg, looks like it was compiled. Now i'm out of home and cannot reboot PVE box.

```
Here is the log:

root@pve:~/test_r8125_kernel61# dpkg -i realtek-r8125-dkms_9.010.01-2_amd64.deb
(Reading database ... 219676 files and directories currently installed.)
Preparing to unpack realtek-r8125-dkms_9.010.01-2_amd64.deb ...

-------- Uninstall Beginning --------
Module:  realtek-r8125
Version: 9.010.01
Kernel:  5.15.74-1-pve (amd64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

r8125.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.15.74-1-pve/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod....

DKMS: uninstall completed.

------------------------------
Deleting module version: 9.010.01
completely from the DKMS tree.
------------------------------
Done.
Unpacking realtek-r8125-dkms (9.010.01-2) over (9.010.01-1) ...
Setting up realtek-r8125-dkms (9.010.01-2) ...
Loading new realtek-r8125-9.010.01 DKMS files...
Building for 5.15.74-1-pve 6.1.0-1-pve
Building for architecture amd64
Building initial module for 5.15.74-1-pve
Done.

r8125.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.15.74-1-pve/updates/dkms/

depmod...

DKMS: install completed.
Building initial module for 6.1.0-1-pve
Done.

r8125.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.1.0-1-pve/updates/dkms/

depmod...

DKMS: install completed.
```

i let you know :)

awesometic, 2022-12-24:
Uploaded the patched version first, but please let me know the results on your side :smiley:

stefanh-it, 2022-12-28:
hi @awesometic 

I ran into the exact same issue after updating to a 6.1 kernel. 

I downloaded and tested your latest version with the applied patch for 6.1 and was able to successfully build the r8125 dkms 

Thank you for taking care of this so quickly!

awesometic, 2023-01-01:
Thank you for letting me know @stefanh-it , I can close this issue then 😄

closed 2023-01-01 by awesometic