awesometic/realtek-r8125-dkms #32: Cannot auto-negotiate the speed to 2.5G in openwrt!

sjkhsl, 2023-01-18:
Cannot auto-negotiate the speed to 2.5G in openwrt!How should it be solved?

```
Supported ports: [ TP ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                2500baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
                                             1000baseT/Full
                                             2500baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Auto-negotiation: on
        Port: Twisted Pair
        PHYAD: 0
        Transceiver: internal
        MDI-X: Unknown
        Supports Wake-on: pumbg
        Wake-on: g
        Current message level: 0x00000033 (51)
                               drv probe ifdown ifup
        Link detected: yes
```

awesometic, 2023-01-19:
Hi,

Does the opposite side device also support 2.5Gbps Ethernet?
I think there is something to do with the network configuration 🤔

sjkhsl, 2023-01-20:
> 你好，
> 
> 对端设备是否也支持2.5Gbps以太网？ 我认为与网络配置有关🤔

The two devices at the opposite end are both 2.5G network cards！

sjkhsl, 2023-01-20:
> Hi,
> 
> Does the opposite side device also support 2.5Gbps Ethernet? I think there is something to do with the network configuration 🤔

The two devices at the opposite end are both 2.5G network cards！

awesometic, 2023-01-26:
Sorry for the late reply,
then is the opposite side device also showing 2500baseT/Full in its available link modes?

sjkhsl, 2023-01-26:
> Sorry for the late reply, then is the opposite side device also showing 2500baseT/Full in its available link modes?

The peer device, a fool switch, is all 2.5G. There is no way to see the rate connection. Only the color of the interface light can distinguish the rate. A light cat can see the 2.5G port! When using the openwrt firmware of lede, 2.5G can be automatically negotiated, and there is no mandatory automatic advertisement in/etc/rc.local. Can't use the official openwrt, so it's probably a firmware problem!

awesometic, 2023-01-26:
So seems like the firmware of the official OpenWrt is the culprit. 🤔 
Then if you think that this realtek-r8125-dkms works normally, please close this issue, or, let me know the further progresses.

closed 2023-01-26 by sjkhsl