awesometic/realtek-r8125-dkms #35: proxmox 7.4 high cpu usage with rtl8125 nic

gpz1100, 2023-04-22:
When doing iperf3 line speed tests, im seeing very high cpu in proxmox host top with the above nic and stock (kernel?) rtl8125 driver.

According to lspci it's using the r8169 driver.

Will installing the driver offered in this github improve with excessive cpu usage?

Thank you

gpz1100, 2023-04-27:
^^Forgot to update.  I did compile the driver and saw a marginal improvement. I don't recall the numbers any longer because it wasn't a huge improvement.

Ultimately I decided to not use the rtl8125 nic in bridge  mode in proxmox. Instead I will pass it through to one of the guest vm's. For bridge mode I am using an intel nic, one of the nics off the i340-t4 card (quad port 82580 based card). Much better cpu and power usage, even at line speeds.

awesometic, 2024-01-22:
Maybe that difference comes from the performance of the controller. I searched that the Intel 82580 supports 1 Gbits, but the Realtek r8125 supports 2.5 Gbits. Can't sure since I didn't do any experiments with them, but makes sense 😄 

I will close this issue, if you are having problem with this package, please make a new issue.

closed 2024-01-22 by awesometic