awesometic/realtek-r8125-dkms #38: Official 23.05 build fails。

sjkhsl, 2023-05-25:
Official 23.05 build fails

```
make[3]: Entering directory '/home/sjkhsl/openwrt-23.05/package/chajianbao/r8125'
touch /home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/.prepared_d201d8b33594c1329ed6c677ebd4c023_6664517399ebbbc92a37c5bb081b5c53_check
true
[ ! -d ./src/ ] || cp -fpR ./src/. /home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00

Applying ./patches/010-config.patch using plaintext:
patching file Makefile

Applying ./patches/020-fix-6.1-suppot.patch using plaintext:
patching file r8125.h

Applying ./patches/100-add-LED-configuration-from-OF.patch using plaintext:
patching file r8125_n.c

Applying ./patches/101-legacy-2500baseX.patch using plaintext:
patching file Makefile
patching file r8125_n.c
touch /home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/.prepared_d201d8b33594c1329ed6c677ebd4c023_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/.configured_*
rm -f /home/sjkhsl/openwrt-23.05/staging_dir/target-x86_64_musl/stamp/.r8125_installed
(cd /home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/./; if [ -x ./configure ]; then find /home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/ -name config.guess | xargs -r chmod u+w; find /home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/ -name config.guess | xargs -r -n1 cp --remove-destination /home/sjkhsl/openwrt-23.05/scripts/config.guess; find /home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/ -name config.sub | xargs -r chmod u+w; find /home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/ -name config.sub | xargs -r -n1 cp --remove-destination /home/sjkhsl/openwrt-23.05/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00=r8125-9.011.00 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00=r8125-9.011.00 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00=r8125-9.011.00 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/sjkhsl/openwrt-23.05/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/usr/include -I/home/sjkhsl/openwrt-23.05/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/include/fortify -I/home/sjkhsl/openwrt-23.05/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/include " LDFLAGS="-L/home/sjkhsl/openwrt-23.05/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/usr/lib -L/home/sjkhsl/openwrt-23.05/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib -znow -zrelro "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/.built
touch /home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/.built_check
make -C /home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.112 KCFLAGS="-fmacro-prefix-map=/home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/sjkhsl/openwrt-23.05/staging_dir/host/include -I/home/sjkhsl/openwrt-23.05/staging_dir/hostpkg/include -I/home/sjkhsl/openwrt-23.05/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sun May 21 17:57:06 2023" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/sjkhsl/openwrt-23.05/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/symvers/button-hotplug.symvers" KERNELRELEASE=5.15.112 M=/home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00 modules
make[4]: Entering directory '/home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.112'
  CC [M]  /home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/r8125_n.o
  CC [M]  /home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/rtl_eeprom.o
  CC [M]  /home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/rtltool.o
  CC [M]  /home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/r8125_rss.o
/home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/r8125_rss.c: In function 'rtl8125_get_rss_hash_opts':
/home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/r8125_rss.c:59:27: error: this statement may fall through [-Werror=implicit-fallthrough=]
   59 |                 cmd->data |= RXH_L4_B_0_1 | RXH_L4_B_2_3;
/home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/r8125_rss.c:61:9: note: here
   61 |         case UDP_V4_FLOW:
      |         ^~~~
/home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/r8125_rss.c:62:20: error: this statement may fall through [-Werror=implicit-fallthrough=]
   62 |                 if (tp->rss_flags & RTL_8125_RSS_FLAG_HASH_UDP_IPV4)
      |                    ^
/home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/r8125_rss.c:65:9: note: here
   65 |         case IPV4_FLOW:
      |         ^~~~
/home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/r8125_rss.c:69:27: error: this statement may fall through [-Werror=implicit-fallthrough=]
   69 |                 cmd->data |= RXH_L4_B_0_1 | RXH_L4_B_2_3;
/home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/r8125_rss.c:71:9: note: here
   71 |         case UDP_V6_FLOW:
      |         ^~~~
/home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/r8125_rss.c:72:20: error: this statement may fall through [-Werror=implicit-fallthrough=]
   72 |                 if (tp->rss_flags & RTL_8125_RSS_FLAG_HASH_UDP_IPV6)
      |                    ^
/home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/r8125_rss.c:75:9: note: here
   75 |         case IPV6_FLOW:
      |         ^~~~
cc1: all warnings being treated as errors
make[5]: *** [scripts/Makefile.build:289: /home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/r8125_rss.o] Error 1
make[4]: *** [Makefile:1906: /home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00] Error 2
make[4]: Leaving directory '/home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.112'
make[3]: *** [Makefile:37: /home/sjkhsl/openwrt-23.05/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.011.00/.built] Error 2
make[3]: Leaving directory '/home/sjkhsl/openwrt-23.05/package/chajianbao/r8125'
time: package/chajianbao/r8125/compile#5.30#0.35#5.64
    ERROR: package/chajianbao/r8125 failed to build.
make[2]: *** [package/Makefile:120: package/chajianbao/r8125/compile] Error 1
make[2]: Leaving directory '/home/sjkhsl/openwrt-23.05'
make[1]: *** [package/Makefile:114: /home/sjkhsl/openwrt-23.05/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sjkhsl/openwrt-23.05'
make: *** [/home/sjkhsl/openwrt-23.05/include/toplevel.mk:231: world] Error 2

```

awesometic, 2023-06-12:
Hello,

Need to patch to disable that `-Wimplicit-fallthrough` warning when compiling the module rather than edit the source codes since there are so many points that can occur that error.

I will make a patch and let you know that.

henrryhant, 2023-06-16:
It has a patch in https://github.com/sbwml/package_kernel_r8125
You can try .