awesometic/realtek-r8125-dkms #39: Driver compiled with PTP support not working

tiz47, 2023-10-18:
Hi,
I'm trying to get hardware timestamp support for nanoPI R5C.
i have edited Makefile changing:
`ENABLE_PTP_SUPPORT = n
ENABLE_PTP_MASTER_MODE = n`
to
`ENABLE_PTP_SUPPORT = y
ENABLE_PTP_MASTER_MODE = y`

Ethtool now indicates that it supports hardware timestamps. However, after less than five minutes of connecting the Ethernet cable, it suddenly stops working. I have to remove and reconnect the cable. 'ip a' reports that the interface is up, and the LED on the NIC is blinking. In 'dmesg,' there are no reports about what happened. (The last log entry is the link-up of the NIC).

Do you have any ideas about where I can find some useful logs, or is the Realtek PTP implementation broken?

awesometic, 2023-10-20:
Hello,

How about `dmesg` returns? If it logs about the error or progress on something, it should appear on that

tiz47, 2023-10-20:
Hi,
thanks for the reply.
This is `dmesg | grep r8125` log from boot:

[   11.403845] r8125: loading out-of-tree module taints kernel.
[   11.404958] r8125: module verification failed: signature and/or required key missing - tainting kernel
[   11.440352] r8125 2.5Gigabit Ethernet driver 9.011.01-NAPI-PTP loaded
[   11.441004] r8125 0001:01:00.0: enabling device (0000 -> 0003)
[   11.483173] r8125 0001:01:00.0 (unnamed net_device) (uninitialized): Invalid ether addr 00:00:00:00:00:00
[   11.484051] r8125 0001:01:00.0 (unnamed net_device) (uninitialized): Random ether addr 46:3b:3f:b2:80:ab
[   11.535208] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[   11.563610] r8125  Copyright (C) 2022 Realtek NIC software team <nicfae@realtek.com> 
[   11.566649] r8125 2.5Gigabit Ethernet driver 9.011.01-NAPI-PTP loaded
[   11.567328] r8125 0002:01:00.0: enabling device (0000 -> 0003)
[   11.568381] r8125 0002:01:00.0: no MSI/MSI-X. Back to INTx.
[   11.584580] r8125 0002:01:00.0 (unnamed net_device) (uninitialized): Invalid ether addr 00:00:00:00:00:00
[   11.585505] r8125 0002:01:00.0 (unnamed net_device) (uninitialized): Random ether addr ea:a0:64:4c:d2:16
[   11.699413] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[   11.702980] r8125  Copyright (C) 2022 Realtek NIC software team <nicfae@realtek.com> 
[   12.227847] r8125 0001:01:00.0 lan0: renamed from eth0
[   12.292567] r8125 0002:01:00.0 wan0: renamed from eth1
[   12.718390] r8125 0001:01:00.0 lan0: registered PHC device on lan0
[   12.719303] r8125 0001:01:00.0 lan0: reset PHC clock
[   12.774125] r8125 0002:01:00.0 wan0: registered PHC device on wan0
[   12.774782] r8125 0002:01:00.0 wan0: reset PHC clock
[   15.111949] r8125 0002:01:00.0: PCIe Bus Error: severity=Corrected, type=Physical Layer, (Transmitter ID)
[   15.113155] r8125 0002:01:00.0:   device [10ec:8125] error status/mask=00001081/0000e000
[   15.114234] r8125 0002:01:00.0:    [ 0] RxErr                  (First)
[   15.115134] r8125 0002:01:00.0:    [ 7] BadDLLP               
[   15.115709] r8125 0002:01:00.0:    [12] Timeout               
[   15.116258] r8125 0002:01:00.0: AER:   Error of this Agent is reported first
[   15.134517] r8125: wan0: link up

Those are the last lines that `dmesg` returns.

[   12.773472] pps pps1: new PPS source ptp1
[   12.774125] r8125 0002:01:00.0 wan0: registered PHC device on wan0
[   12.774782] r8125 0002:01:00.0 wan0: reset PHC clock
[   15.107465] pcieport 0002:00:00.0: AER: Multiple Corrected error received: 0002:01:00.0
[   15.108620] pcieport 0002:00:00.0: PCIe Bus Error: severity=Corrected, type=Physical Layer, (Receiver ID)
[   15.109850] pcieport 0002:00:00.0:   device [1d87:3566] error status/mask=00000001/0000e000
[   15.111011] pcieport 0002:00:00.0:    [ 0] RxErr                  (First)
[   15.111949] r8125 0002:01:00.0: PCIe Bus Error: severity=Corrected, type=Physical Layer, (Transmitter ID)
[   15.113155] r8125 0002:01:00.0:   device [10ec:8125] error status/mask=00001081/0000e000
[   15.114234] r8125 0002:01:00.0:    [ 0] RxErr                  (First)
[   15.115134] r8125 0002:01:00.0:    [ 7] BadDLLP               
[   15.115709] r8125 0002:01:00.0:    [12] Timeout               
[   15.116258] r8125 0002:01:00.0: AER:   Error of this Agent is reported first
[   15.134517] r8125: wan0: link up
[   15.135092] IPv6: ADDRCONF(NETDEV_CHANGE): wan0: link becomes ready

I checked by connecting a monitor directly to the nanopi and nothing else is written in `dmesg` when the problem occurs.
I disconnect and reconnect the cable, `dmesg` only returns to the disconnection and reconnection:
`[  153.683275] r8125: wan0: link down
[  156.315009] r8125: wan0: link up`

awesometic, 2023-10-25:
Well, on the whole there are three possibilities.

1. Fault on the driver - The driver should do all its functions by the flags without issue
2. Fault on the kernel - For some reason the driver can be built with NanoPi's kernel but it can have issues when the implementation is different from the driver expects
3. Fault on NanoPi - For some reason it can have a circuit issue

So, yes, it is hard to define which part has the issue but you can give it a try to contact NanoPi forum first.
If the same issue exists on the x86 PC, then the fault is on the driver.

tiz47, 2023-11-07:
Hi,
with the latest version (9.012.03) it seems to work correctly.

closed 2023-11-07 by tiz47