awesometic/realtek-r8125-dkms #42: Very good Makefile. Make my kernel upgrade failed

BoringCat, 2023-11-29:
dkms use `'make' -j16 KVER=x.x.x-xxxxxx BSRC=/lib/modules/x.x.x-xxxxxx` to run Makefile. please use KVER instead of `uname -r`

here is my diff file
```diff
diff --git a/Makefile b/Makefile
index 3aa0954..161fde3 100755
--- a/Makefile
+++ b/Makefile
@@ -30,26 +30,27 @@
 #  US6,570,884, US6,115,776, and US6,327,625.
 ################################################################################
 
-KFLAG := 2$(shell uname -r | sed -ne 's/^2\.[4]\..*/4/p')x
+KVER ?= 2$(shell uname -r | sed -ne 's/^2\.[4]\..*/4/p')x
+export KVER = ${KFLAG}
 
 all: clean modules install
 
 modules:
-ifeq ($(KFLAG),24x)
+ifeq ($(KVER),24x)
        $(MAKE) -C src/ -f Makefile_linux24x modules
 else
        $(MAKE) -C src/ modules
 endif
 
 clean:
-ifeq ($(KFLAG),24x)
+ifeq ($(KVER),24x)
        $(MAKE) -C src/ -f Makefile_linux24x clean
 else
        $(MAKE) -C src/ clean
 endif
 
 install:
-ifeq ($(KFLAG),24x)
+ifeq ($(KVER),24x)
        $(MAKE) -C src/ -f Makefile_linux24x install
 else
        $(MAKE) -C src/ install
diff --git a/src/Makefile b/src/Makefile
index 5175cf2..483360b 100755
--- a/src/Makefile
+++ b/src/Makefile
@@ -132,7 +132,8 @@ ifneq ($(KERNELRELEASE),)
                EXTRA_CFLAGS += -DENABLE_RX_PACKET_FRAGMENT
        endif
 else
-       BASEDIR := /lib/modules/$(shell uname -r)
+       KVER ?= $(shell uname -r)
+       BASEDIR ?= /lib/modules/${KVER}
        KERNELDIR ?= $(BASEDIR)/build
        PWD :=$(shell pwd)
        DRIVERDIR := $(shell find $(BASEDIR)/kernel/drivers/net/ethernet -name realtek -type d)
@@ -147,7 +148,6 @@ else
        KERNEL_GCC_VERSION := $(shell cat /proc/version | sed -n 's/.*gcc version \([[:digit:]]\.[[:digit:]]\.[[:digit:]]\).*/\1/p')
        CCVERSION = $(shell $(CC) -dumpversion)
 
-       KVER = $(shell uname -r)
        KMAJ = $(shell echo $(KVER) | \
        sed -e 's/^\([0-9][0-9]*\)\.[0-9][0-9]*\.[0-9][0-9]*.*/\1/')
        KMIN = $(shell echo $(KVER) | \
```
> test on archlinux. upgraded linux (6.5.9.arch2-1 -> 6.6.2.arch1-1)

awesometic, 2023-11-30:
Thanks for the info. Someone here also suffered from this issue. 😅
I will check the diffs by you and this pull request and accept one or both

https://github.com/awesometic/realtek-r8125-dkms/pull/41

awesometic, 2023-12-01:
Since your diffs includes editing `src/Makefile`, I'd like to find another way to resolve this issue. Editing files under `src` makes it more difficult to apply the upcoming versions.

Let me check, and will update the make process within this weekend. Sorry for the inconvenience 😅

dominicx, 2024-01-16:
> Since your diffs includes editing `src/Makefile`, I'd like to find another way to resolve this issue. Editing files under `src` makes it more difficult to apply the upcoming versions.
> 
> Let me check, and will update the make process within this weekend. Sorry for the inconvenience 😅

I also modified src/Makefile when I upgraded kernel 5.x to 6.2, The pull #41 I submitted could also solve this problem using the BASEDIR  overriding assignment on dkms.conf. 
I noticed that you have already merged that pull. 
Thank you for your great work.

dominicx, 2024-01-16:
KVER on src/Makefile override by 'KVER=${kernelver}' on dkms.conf .
Finally, it is overwritten by the assignment of the dkms command parameter -k.

awesometic, 2024-01-22:
Yeah, the patch can solve the issue this thread originally mentioned.

I will close this. If there's another issue that comes up, please make a new issue.

closed 2024-01-22 by awesometic