awesometic/realtek-r8125-dkms #44: Module not loading on latest Linux Mint/Ubuntu kernel 6.5

mikeNG, 2023-12-18:
```
$ dmesg | grep r8125
[    6.348755] module r8125: .gnu.linkonce.this_module section size must match the kernel's built struct module size at run time
```

```
$ uname -a
Linux kraken 6.5.0-14-generic #14~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Nov 20 18:15:30 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

awesometic, 2023-12-18:
Hello,

Looks like something changed by this commit recently,

- https://lore.kernel.org/lkml/ZBuShv1hx03WmzH4@bombadil.infradead.org/T/#maf36ff06626a1477f652908e77dc15387ffc335b

And it seems this also makes the developers something more on their module but I am not sure.

I will investigate and patch our project 🤔

awesometic, 2024-01-21:
Late update, but recently I updated my server to kernel 6.5 it just works... after some fixing.

```
$ uname -a
Linux awesometic-nas 6.5.0-14-generic #14~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Nov 20 18:15:30 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

Right after the kernel update and reboot, the module wasn't loaded and I found that the `vermagic` is not updated to the new kernel version. 

I thought that is not compiled in the kernel updating scripts, and I gave a shot to reinstall this package using `apt`.

```
$ modinfo r8125 | grep vermagic
vermagic:       6.5.0-14-generic SMP preempt mod_unload modversions
```

And reboot again, and it works again.

Just a note for the other people, I don't think I had the same problem with you because I couldn't see any message on `dmesg` when the ethernet didn't work.

barkmann44, 2024-01-26:
Hello awesometic, I am also having this problem with your latest release, module no longer loads "Right after the kernel update and reboot, " Ubuntu server 22.04.  So I reinstalled module using sudo dpkg -i, however the module no longer loads after a new kernel is installed.  Can you please advise how to fix?  Should I reinstall using apt? Many thanks!!

awesometic, 2024-01-26:
Hello @barkmann44, in my case I can fix it by reinstalling the Debian package.

```
sudo apt install ./realtek-r8125-dkms_9.012.04-1_amd64.deb --reinstall
```

Is `vermagic` correct to your new kernel? If so, is there any message in `dmesg`?

awesometic, 2024-01-26:
Seems like kernel 6.5 breaks so many DKMS friends :)

- https://www.google.com/search?q=kernel+6.5+dkms

barkmann44, 2024-01-26:
Dear awsometic, thanks for responding!

Prior to running apt install, I get this:

vermagic:       5.15.0-92-generic SMP mod_unload modversions

Then trying to install via apt:

chris@hal:~/downloads$ sudo apt install ./realtek-r8125-dkms_9.012.04-1_amd64.deb --reinstall
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'realtek-r8125-dkms' instead of './realtek-r8125-dkms_9.012.04-1_amd64.deb'
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 2 not upgraded.
Need to get 0 B/121 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 /home/chris/downloads/realtek-r8125-dkms_9.012.04-1_amd64.deb realtek-r8125-dkms amd64 9.012.04-1 [121 kB]
(Reading database ... 141823 files and directories currently installed.)
Preparing to unpack .../realtek-r8125-dkms_9.012.04-1_amd64.deb ...
Module realtek-r8125-9.012.04 for kernel 5.15.0-92-generic (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

r8125.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.15.0-92-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod....
Deleting module realtek-r8125-9.012.04 completely from the DKMS tree.
Unpacking realtek-r8125-dkms (9.012.04-1) over (9.012.04-1) ...
Setting up realtek-r8125-dkms (9.012.04-1) ...
Loading new realtek-r8125-9.012.04 DKMS files...
Building for 5.15.0-92-generic
Building for architecture x86_64
Building initial module for 5.15.0-92-generic
Secure Boot not enabled on this system.
Done.

r8125.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.15.0-92-generic/updates/dkms/

depmod...
Scanning processes...
Scanning processor microcode...
Scanning linux images...

Running kernel seems to be up-to-date.

The processor microcode seems to be up-to-date.

No services need to be restarted.

No containers need to be restarted.

No user sessions are running outdated binaries.

No VM guests are running outdated hypervisor (qemu) binaries on this host.
N: Download is performed unsandboxed as root as file '/home/chris/downloads/realtek-r8125-dkms_9.012.04-1_amd64.deb     ' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
chris@hal:~$ modinfo r8125 | grep vermagic
vermagic:       5.15.0-92-generic SMP mod_unload modversions

So.. It doesn't seem to have changed anything regarding later kernel compatibility, I see there is a permission denied message for apt.   Any ideas please?  Thankyou in advance!

awesometic, 2024-01-26:
Okay, I get it now. Maybe it is another problem with the Debian package. 🤔 

Let me have some time to look into this problem. :)

awesometic, 2024-01-26:
Lastly, could you send me the results of this command?

```
sudo dmesg | grep -i r8125
```

gpny2791, 2024-01-26:
Hello, since the last update, i lost my connection with Device-2: Realtek RTL8125 2.5GbE 
The result  of  sudo dmesg | grep -i r8125:
    1.172450] r8125: disagrees about version of symbol module_layout
[    5.755149] r8125: disagrees about version of symbol module_layout

gpny2791, 2024-01-26:
and the result of  "insmod: ERROR: could not insert module /lib/modules/5.15.0-92-generic/updates/dkms/r8125.ko": Invalid module format
insmod: ERROR: could not insert module /lib/modules/5.15.0-92-generic/updates/dkms/r8125.ko: Invalid module format

gpny2791, 2024-01-26:
and 
sudo dmesg | grep -i r8125
[    1.172450] r8125: disagrees about version of symbol module_layout
[    5.755149] r8125: disagrees about version of symbol module_layout
[ 7972.924339] r8125: disagrees about version of symbol module_layout

barkmann44, 2024-01-26:
chris@hal:~$ sudo dmesg | grep -i r8125
[    4.099514] r8125: loading out-of-tree module taints kernel.
[    4.105578] r8125: module verification failed: signature and/or required key missing - tainting kernel
[    4.111729] r8125 Ethernet controller driver 9.012.04-NAPI loaded
[    4.111770] r8125 0000:01:00.0: enabling device (0000 -> 0003)
[    4.136909] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[    4.140937] r8125  Copyright (C) 2023 Realtek NIC software team <nicfae@realtek.com>
[    4.314590] r8125 0000:01:00.0 enp1s0: renamed from eth0
[    7.595215] r8125: enp1s0: link up

chris@hal:~$ sudo insmod /lib/modules/$(uname -r)/updates/dkms/r8125.ko
insmod: ERROR: could not insert module /lib/modules/5.15.0-92-generic/updates/dkms/r8125.ko: File exists

thanks for looking into this awesometic!

awesometic, 2024-04-22:
Maybe it is related to this PR: https://github.com/awesometic/realtek-r8125-dkms/pull/52

Please check this and try with the new Debian package from here: https://github.com/awesometic/realtek-r8125-dkms/releases/tag/9.013.02-1
Or PPA.

mikeNG, 2024-04-22:
> Maybe it is related to this PR: #52
> 
> Please check this and try with the new Debian package from here: https://github.com/awesometic/realtek-r8125-dkms/releases/tag/9.013.02-2 Or PPA.

That has indeed resolved the issue, thanks!

closed 2024-04-22 by mikeNG