awesometic/realtek-r8125-dkms #45: Error: r8125: module verification failed: signature and/or required key missing - tainting kernel

logopk, 2024-01-02:
Building this on ProxMox 8.1.3 (Debian 12) leads to the following message:

```
r8125                 307200  0
[    0.364872] pci 0000:03:00.0: [10ec:8125] type 00 class 0x020000
[    0.942986] r8169 0000:03:00.0 eth0: RTL8125B, a8:a1:59:52:38:c5, XID 641, IRQ 129
[43228.024646] r8125: module verification failed: signature and/or required key missing - tainting kernel
```

Any idea what this means?

Thank you

Peter

----

Installing using this:
```
version=9.012.04-1
curl -L https://github.com/awesometic/realtek-r8125-dkms/archive/refs/tags/${version}.tar.gz -o ${version}.tar.gz
tar xvzf ${version}.tar.gz
cp -pr realtek-r8125-dkms-${version}* /usr/src/r8125-${version}

dkms add -m r8125 -v $version
dkms build -m r8125 -v $version
dkms install -m r8125 -v $version
depmod -a
update-initramfs -u
modprobe -v r8125 

lsmod | grep 8125
dmesg | grep -i 8125
```

awesometic, 2024-01-11:
Hello, is r8169 module also loaded?

Don't remember loading it over r8169 causes that error, but it is still worth trying since it should be loaded alone. please check the following contents.

- https://github.com/awesometic/realtek-r8125-dkms?tab=readme-ov-file#verify-the-module-is-loaded-successfully

logopk, 2024-01-11:
Apparently all is working. The message

`[43228.024646] r8125: module verification failed: signature and/or required key missing - tainting kernel`

is new since kernel 6 and from what I've read only informative.

closed 2024-01-11 by logopk