awesometic/realtek-r8125-dkms #46: Unable to locate package realtek-r1825-dkms

AlbertsSons, 2024-02-26:
I'm trying to install via the Launchpad PPA and I'm getting the subject error.
```
root@:-#` add-apt-repository ppa:awesometic/ppa
 Glad to see you.

With this PPA you can install the following packages.
- realtek-r8125-dkms
- realtek-r8152-dkms

If there's an issue during using my packages, please let me know using the Github repository.
- https://github.com/awesometic
You can see the repositories for each package.

Thank you :)
 More info: https://launchpad.net/~awesometic/+archive/ubuntu/ppa
Press [ENTER] to continue or ctrl-c to cancel adding it

gpg: keybox '/tmp/tmp_uuag26b/pubring.gpg' created
gpg: /tmp/tmp_uuag26b/trustdb.gpg: trustdb created
gpg: key B44E6C80FF7107B6: public key "Launchpad PPA for Yang Deokgyu" imported
gpg: Total number processed: 1
gpg:               imported: 1
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
gpg: no valid OpenPGP data found.

root@:~# apt install realtek-r8125-dkms
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package realtek-r8125-dkms
```

awesometic, 2024-02-26:
Hello,

What is your Linux distribution? Can you show me the results of `cat /etc/issue`?

awesometic, 2024-02-26:
I guess you're using Ubuntu 23.10 or up, I will add support for Mantic (23.10) and Noble (24.04) soon.

AlbertsSons, 2024-02-26:
Hello,  I'm using openmediavault which I believe is debian based.
Thankyou
```
cat /etc/issue
openmediavault 6.9.14-1 (Shaitan) \n \l
Copyright (C) 2009-2024 by Volker Theile. All rights reserved.

To manage the system visit the openmediavault workbench:

br-26dbbdfaaa6e: ###.###.###.###
br-26dbbdfaaa6e: ####::####:####:####:####
docker0: ###.###.###.###
eno1: ###.###.###.###
enp1s0: ###.###.###.###
vethc7292d6f: ###.###.###.###
vethc7292d6f: ####::####:####:####:####
vethd51c8f0: ####::####:####:####:####
wgnet1: ###.###.###.###

For more information regarding this appliance, please visit the
web site: https://www.openmediavault.org
```

AlbertsSons, 2024-02-26:
In case it helpful:
```
uname -r
6.1.0-0.deb11.13-amd64
```

AlbertsSons, 2024-02-26:
Thanks, I'll test it when you add the new support.

awesometic, 2024-03-01:
Hello,

Now Launchpad builds it for Ubuntu Mantic (23.10) and Noble (24.04) too. Please update your package manager and try it again.

```
apt update
apt install realtek-r8125-dkms
```

AlbertsSons, 2024-03-01:
```
Err:14 http://ppa.launchpad.net/awesometic/ppa/ubuntu noble InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY B44E6C80FF7107B6
Reading package lists... Done
W: GPG error: http://ppa.launchpad.net/awesometic/ppa/ubuntu noble InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY B44E6C80FF7107B6
E: The repository 'http://ppa.launchpad.net/awesometic/ppa/ubuntu noble InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
root@openmediavault:~# apt install realtek-r8125-dkms
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package realtek-r8125-dkms
```

AlbertsSons, 2024-03-01:
```
apt update --allow-insecure-repositories
...
W: The repository 'http://ppa.launchpad.net/awesometic/ppa/ubuntu noble InRelease' is not signed.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
root@openmediavault:~# apt install realtek-r8125-dkms
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  realtek-r8125-dkms
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
Need to get 121 kB of archives.
After this operation, 1,091 kB of additional disk space will be used.
WARNING: The following packages cannot be authenticated!
  realtek-r8125-dkms
Install these packages without verification? [y/N] y
Get:1 http://ppa.launchpad.net/awesometic/ppa/ubuntu noble/main amd64 realtek-r8125-dkms amd64 9.012.04-1~202403010811~ubuntu24.04.1 [121 kB]
Fetched 121 kB in 1s (223 kB/s)
Selecting previously unselected package realtek-r8125-dkms.
(Reading database ... 65076 files and directories currently installed.)
Preparing to unpack .../realtek-r8125-dkms_9.012.04-1~202403010811~ubuntu24.04.1_amd64.deb ...
Unpacking realtek-r8125-dkms (9.012.04-1~202403010811~ubuntu24.04.1) ...
Setting up realtek-r8125-dkms (9.012.04-1~202403010811~ubuntu24.04.1) ...
Loading new realtek-r8125-9.012.04 DKMS files...
Building for 6.1.0-0.deb11.13-amd64
Building for architecture amd64
Module build for kernel 6.1.0-0.deb11.13-amd64 was skipped since the
kernel headers for this kernel does not seem to be installed.
```

AlbertsSons, 2024-03-01:
I'm not sure how to fix the issue with the kernel headers.  I believe I have them installed but when it tries to build the realtek drivers it says the headers don't appear to be installed.

awesometic, 2024-03-02:
Maybe you can find a way to fix it by searching about DKMS on OMV 6 🤔

AlbertsSons, 2024-03-02:
Thanks, I will.  I'll update this post if I figure it out.

AlbertsSons, 2024-03-07:
I've not fixed this yet but for anyone arriving here with a similar kernel header issue I've had discussions on [reddit](https://old.reddit.com/r/linux4noobs/comments/1b5q8wa/dkms_and_kernel_headers/) and the [OMV forums](https://forum.openmediavault.org/index.php?thread/51628-dkms-and-kernel-headers/).

AlbertsSons, 2024-03-07:
I'm closing this issue since, as related to the subject specifically, I was able to locate the package.

closed 2024-03-07 by AlbertsSons