awesometic/realtek-r8125-dkms #48: What's the advantage of using realtek driver instead of mainline r8169?

amazingfate, 2024-03-05:
Hello, I am a developer from armbian and I'm handling arm SBCs with RTL8125 ethernet.
You started this project from 2020 and I guess mainline r8169 should have some bugs or features not implemented. Would you please tell me why you choose realtek driver instead of mainline r8169 since most distros just ship the mainline driver.

I am thinking of whether using realtek driver in kernel 6.1 for rockchip SBCs. If the reason you choose realtek driver is fixed in kernel 6.1, I may just use mainline r8169 for RTL8125.

Many thanks!

awesometic, 2024-03-05:
Glad to see you here, I love Armbian as a single-board computer lover. It is really well made OS for general use with SBC.

At that moment, 2020, the r8169 module doesn't support r8125 at all. Initially, I developed this package for Odroid-H2 which has two LAN ports that use RTL8125. Yes, as far as I know H2 is the first SBC that has RTL8125, and at that time there is no (maybe a very few) mainboard has this chipset.

So, to use Ethernet on the machine, it must have Realtek one since there is no dedicated module or firmware to use this RTL8125. If my remembering is right, r8169 starts to support RTL8125 in 2021 or 2022. Nowadays, r8169 supports RTL8125 natively so that actually we don't need to install this one by hand for general use.

Then right, why we're using it is not that noticeable. I didn't benchmark in a hard condition between r8169 in the mainline kernel and r8125 from Realtek. Both does work very well with full speed.

But with r8125, we can adjustment the props before compiling the module, and it surely has the latest changes from Realtek. As you know, it is very slow to apply the changes to the mainline since it requires well structured codes and descriptions, and reviews from multiple people.
One of the latest changes is to support 5GbE (RTL8126) that is not popular in the mainboard world yet.

Plus, I have added the supports for the newer kernel with this project and have fixed some bugs on some conditions. You can check the commit history of that. Now the source from Realtek doesn't need my patches, but I don't know, maybe they're watching this repository too.

I think you can choose r8169 for the best stability. But you may choose r8125 for some additional features and tunes. Sorry, I can't tell you the exact difference in features in number but there should be a reason that Realtek is publishing the dedicated driver on their website.

I hope this helps you to choose one for your amazing work. :)

amazingfate, 2024-03-05:
Thanks for your reply! I'm working on a kernel 6.1 released in 2022 so I think armbian can ship the mainline r8169 driver and users can use dkms for realtek driver by their own decision.

closed 2024-03-05 by amazingfate

rpardini, 2024-03-06:
Happy to report @awesometic's awesometic/realtek-r8125-dkms builds on Armbian, using the new Rockchip 'vendor' 6.1-rkr1 kernel by @amazingfate & co, and it's linux-headers package (it should also build with the 5.10-rkr6 `legacy` kernel):

```bash
apt install dkms git linux-headers-vendor-rk35xx
git clone https://github.com/awesometic/realtek-r8125-dkms.git
cd realtek-r8125-dkms/
bash dkms-install.sh
```