awesometic/realtek-r8125-dkms #49: Driver not loaded on PVE 8.1.4

kkjj5566123, 2024-03-26:
I followed the Installation and successfully installed driver with Launchpad PPA and blacklisted r8169, but after I rebooted and use lspci, It turns out no kernel driver in use, any idea to resolve this?
root@Main:~# lspci -s 02:00.0 -k
02:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
        Subsystem: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller
        Kernel modules: r8169, r8125

LsnmxNB, 2024-03-26:
## 安装headers
$ headers=$(dpkg -l | awk '/^ii.+kernel-[0-9]+\.[0-9]+\.[0-9]/{gsub(/-signed/, ""); gsub(/kernel/, "headers"); print $2}' | tr "\n" " ")
$ eval apt install -y $headers

## 安装刚刚下载好的deb包，此命令只会为当前系统所使用的内核以及刚刚安装的最新内核（也可能当前所使用的内核就是最新内核）安装驱动
$ dpkg -i realtek-r8125-dkms_*.deb  # 如通配符会匹配多个时亦可指定具体的文件名

## 如果想为那些既不是系统当前所使用的内核，也不是刚刚安装的最新内核安装驱动，需要手动指定安装
## 看看哪些内核安装好了驱动
$ dkms status
## 列出全部内核版本kernel_version，找出那些还没有安装驱动的内核
$ dpkg -l | awk '/^ii.+kernel-[0-9]+\.[0-9]+\.[0-9]/{gsub(/proxmox-kernel-|pve-kernel-|-signed/, ""); print $2}'
## 手动指定驱动版本（在deb文件名中有体现）和内核版本（从上一句命令的输出中），zsh按tab可自动补全，比如：dkms install realtek-r8125/9.011.01 -k 6.2.16-5-pve
$ dkms install realtek-r8125/<driver_version> -k <kernel_version>

## 卸载不再需要的headers
$ eval apt-mark auto $headers
$ apt autopurge

root@LSNMX:~# dpkg -l | awk '/^ii.+kernel-[0-9]+\.[0-9]+\.[0-9]/{gsub(/proxmox-kernel-|pve-kernel-|-signed/, ""); print $2}'
6.5.11-8-pve
6.5.13-3-pve
root@LSNMX:~# dkms status
realtek-r8125/9.012.04, 6.5.13-3-pve, x86_64: installed
root@LSNMX:~# 

root@LSNMX:~# grep -E enp* /proc/interrupts
 135:     153413          0          0          0  IR-PCI-MSIX-0000:02:00.0    0-edge      enp2s0-0
 136:          0     195520          0          0  IR-PCI-MSIX-0000:02:00.0    1-edge      enp2s0-1
 137:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0    2-edge      enp2s0-2
 138:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0    3-edge      enp2s0-3
 139:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0    4-edge      enp2s0-4
 140:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0    5-edge      enp2s0-5
 141:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0    6-edge      enp2s0-6
 142:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0    7-edge      enp2s0-7
 143:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0    8-edge      enp2s0-8
 144:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0    9-edge      enp2s0-9
 145:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0   10-edge      enp2s0-10
 146:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0   11-edge      enp2s0-11
 147:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0   12-edge      enp2s0-12
 148:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0   13-edge      enp2s0-13
 149:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0   14-edge      enp2s0-14
 150:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0   15-edge      enp2s0-15
 151:     297277          0          0          0  IR-PCI-MSIX-0000:02:00.0   16-edge      enp2s0-16
 152:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0   17-edge      enp2s0-17
 153:          0          0     224917          0  IR-PCI-MSIX-0000:02:00.0   18-edge      enp2s0-18
 154:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0   19-edge      enp2s0-19
 155:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0   20-edge      enp2s0-20
 156:          0        599          0          0  IR-PCI-MSIX-0000:02:00.0   21-edge      enp2s0-21
 157:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0   22-edge      enp2s0-22
 158:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0   23-edge      enp2s0-23
 159:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0   24-edge      enp2s0-24
 160:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0   25-edge      enp2s0-25
 161:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0   26-edge      enp2s0-26
 162:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0   27-edge      enp2s0-27
 163:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0   28-edge      enp2s0-28
 164:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0   29-edge      enp2s0-29
 165:          0          0          0          0  IR-PCI-MSIX-0000:02:00.0   30-edge      enp2s0-30

LsnmxNB, 2024-03-26:
https://evine.win/p/pve-install-realtek-8125-driver/

closed 2024-04-14 by kkjj5566123