awesometic/realtek-r8125-dkms #51: Kernel update breaks network

Br33ce, 2024-04-14:
Hey and thank you for this awesome repo. I am running Debian 12 (Openmediavault 7 with a Gigabyte B760m MB with a RTL8125 network chip) and just upgrade the kernel from 6.1.0-18 to 20 (https://mirrors.edge.kernel.org/pub/linux/kernel/v6.x/ChangeLog-6.1.83)). After reboot my network connection is completely gone and the network interface is not found. 
On the older kernel your driver install worked flawlessly and I dont know what went wrong here.
I think something went wrong with DKMS here:

`Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/realtek-r8125/9.012.04/source/dkms.conf)

Building module:
Cleaning build area...
'make' -j8 KVER=6.1.0-20-amd64 BASEDIR=/lib/modules/6.1.0-20-amd64 modules.....
Signing module /var/lib/dkms/realtek-r8125/9.012.04/build/src/r8125.ko
Cleaning build area...

r8125.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.1.0-20-amd64/updates/dkms/
depmod...
dkms: autoinstall for kernel: 6.1.0-20-amd64.
/etc/kernel/postinst.d/initramfs-tools:
update-initramfs: Generating /boot/initrd.img-6.1.0-20-amd64
/etc/kernel/postinst.d/zz-update-grub:
Generating grub configuration file ...
Found linux image: /boot/vmlinuz-6.1.0-20-amd64
Found initrd image: /boot/initrd.img-6.1.0-20-amd64
Found linux image: /boot/vmlinuz-6.1.0-18-amd64
Found initrd image: /boot/initrd.img-6.1.0-18-amd64
Found memtest86+ 64bit EFI image: /boot/memtest86+x64.efi
Warning: os-prober will not be executed to detect other bootable partitions.
Systems on them will not be added to the GRUB boot configuration.
Check GRUB_DISABLE_OS_PROBER documentation entry.
Adding boot menu entry for UEFI Firmware Settings ...
done
Setting up linux-headers-6.1.0-20-common (6.1.85-1) ...
Setting up linux-image-amd64 (6.1.85-1) ...
Setting up linux-headers-6.1.0-20-amd64 (6.1.85-1) ...
/etc/kernel/header_postinst.d/dkms:
dkms: running auto installation service for kernel 6.1.0-20-amd64.
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/realtek-r8125/9.012.04/source/dkms.conf)
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/realtek-r8125/9.012.04/source/dkms.conf)
dkms: autoinstall for kernel: 6.1.0-20-amd64.
Setting up linux-headers-amd64 (6.1.85-1) ...`

awesometic, 2024-04-17:
Hello,

I think It is very weird. I need to test it on my side by installing another kernel.
In the logs DKMS auto installation worked well but seems like for some reason the latest version of the r8125 module wasn't loaded.

Am I missing something on the logs? 🤔

awesometic, 2024-04-18:
By the way, how did you install the package? `apt` or the Debian package file from here?

Br33ce, 2024-04-18:
Hey,

thank you very much for your reply. I've used the deb file method. With apt you mean launchpad, right?
I've checked in with the folks at Openmediavault but have not gotten much of a reply "A new kernel means you need to recompile all drivers". That's what DKMS was supposed to do, wasn't it?
I did manage to fix it but am not 100% what did it in the end.
First I checked at realtek and there is a newer driver version available. I tried to install it with the new kernel, which failed, and the old kernel (which I can still boot). That did not do anything. I then tried to figure out if maybe the name of my ethernet device changed, I've read that happens sometimes, and so the new kernel does not recognize it.
The name did change from enp4s0 to enp5...
I then did `sudo update-initramfs -u` and now the system works with the new kernel.

I don't know if this helps you with troubleshooting. If not, please tell me what you need.

awesometic, 2024-04-26:
The network interface name can be changed and can be set by a manager like NetworkManager from the device name showing like 'eth*'. Maybe this can make it won't happen anymore?

Thank you for sharing your experience, I will close this issue. 😃

closed 2024-04-26 by awesometic

Presjar, 2025-01-26:
The driver is no longer loaded for me either whenever Proxmox updates the kernel. This has happened several times in the last 6 months. I'm guessing every kernel change.

To fix this I have been accessing the server using the usually unused Intel NIC to download the required headers and then reinstall the deb package.

Is this as expected or is something incorrect with my setup?

Br33ce, 2025-01-27:
I can only add, that with my system kernel updates now work as intendend with the latest version.