awesometic/realtek-r8125-dkms #55: 9.013.02 not work on pve 8.2 with kernel 6.8

Vohrt, 2024-05-10:
PVE 8.2 changes their default kernel to 6.8, which makes my network down. After 
upgrading r8125-dkms to 9.013.02, there is still lots of errors show on my console, like:

 [353.459395] r8125 0000:06:00.0 enp6s0: NETDEV WATCHDOG: CPU: 0: transmit queue 0 timed out 12412 ms

and the network is still unavailable.

I'm not familiar with dkms so i don't know how to get more logs about this error. If you need more info, just tell me.

By the way, the newset one still works on pve 8.2 with kernel 6.5.

awesometic, 2024-05-16:
Sorry for the inconvenience. I will test it on my kernel 6.8 system soon.

awesometic, 2024-05-16:
How did you upgrade the package? Using Debian package file or the script the repo includes?

Vohrt, 2024-05-16:
> How did you upgrade the package? Using Debian package file or the script the repo includes?

Using Debian package file with dpkg -i.

awesometic, 2024-05-17:
Thanks @whidy for sharing your experience.

Although the driver can be downloaded directly from the Realtek website, the source code in this repository is identical to the official one. If you're curious about the previously merged patches to the sources here, they have been merged on the official side already.

So you can install the driver without DKMS by cloning this repository to your PVE machine and executing the `autorun.sh` script. It should work.

But it is always your choice, you may be relieved by downloading the source codes from the official website.

AndyGaskell, 2024-06-11:
I'm not sure if this is the same issue, but I'm using Ubuntu 24.04, which comes with the 6.8 kernel.

When I install the package via the launchpadd ppa, as per the readme, and blacklisted the r8169.

So lspci -k gives...
`01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
	Subsystem: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller
	Kernel driver in use: r8125
	Kernel modules: r8169, r8125
`

But as a connection it's barely functioning, dropping 70+% of packets. 

The cable and gig-e switch it is plugged into are4 all fine.

This is with a MINIX Z100-0dB fanless Intel N100 mini PC.

awesometic, 2024-06-12:
@AndyGaskell 

How about r8196? If it has the same issue (high rate packet loss), it might be a hardware issue.

wknd, 2024-06-22:
I can confirm the `.deb` package does work for me on proxmox 8.

I just tried it on 2 identical machines with a fresh install of proxmox 8 and it works the same as using the driver from the realtek website manually.

order of operations for success:
- fresh install of proxmox 8
- update package repositories to non licensed
- update and upgrade packages
- install dependencies: `apt install build-essential pve-headers proxmox-default-headers dkms`
- blacklist evil driver by placing `blacklist r8169` in `/etc/modprobe.d/realtekblacklist.conf`
- download package: `wget https://github.com/awesometic/realtek-r8125-dkms/releases/download/9.013.02-2/realtek-r8125-dkms_9.013.02-2_amd64.deb`
- install: `dpkg -i realtek-r8125-dkms`
- change network interface to the now working one in `/etc/network/interfaces`
- rebuild initramfs just for fun (probably not needed since we blacklisted before installing?): `update-initramfs -u`
- reboot

The most important proxmox specific part is probably first installing `pve-headers` and `proxmox-default-headers` **before** trying to install the `.deb` package.

Hopefully this means next kernel update it will automagically rebuild and continue working compared to manually building from the realtek website.

Vohrt, 2024-12-29:
After half a year, I come back to this post to add some information, because I found that many other users pointed to this post
[](https://forum.proxmox.com/threads/realtek-nic-rtl8125bg-driver-issues.151566/)
[](https://forum.proxmox.com/threads/4port-network-card-not-recognized.159653/)

First of all, I still haven't solved this problem. I am still insisting on using kernel 6.5, and I can only hope that pve will not give up support for version 6.5 for the time being. I tried the solutions provided by several friends above, but even if it is a new installation, the same error will still occur when entering the same commands in sequence.

By the way, I also tried to use the latest beta version 6.11 kernel of pve, and the problem I encountered was the same.

According to the current feedback, the problem mainly occurs on the third-party 4-port RTL8125 [BG], and the version information displayed in lspci is mostly [rev 05]. These network cards work fine under Windows, so there is no hardware problem.

The current speculation may be 1) the problem caused by PVE's modification of the kernel, 2) the problem caused by the Linux kernel itself, 3) the problem caused by the Linux driver provided by Realtek not considering the special hardware of the multi-port network card of the [BG] version

For users who find this later, because the newly installed PVE8.2 has the 6.8 kernel by default, if you want to temporarily solve the problem, you can try to manually install the 6.5 kernel and select the 6.5 kernel when booting.

If your device has a network card other than 8125, try to connect to the Internet with it first, then enter:

apt install proxmox-kernel-6.5

After the installation is complete, use the up and down keys on the keyboard to select the advanced... option on the blue screen at startup, and then select with kernel 6.5... After entering the system, follow the methods of the above friends to install the driver, either using dkms or directly using the source code. Note that if you want to enter the 6.5 kernel every time you boot, you need to google the settings, otherwise the system will return to the 6.8 kernel after the next restart.

firebat81, 2025-01-02:
This error also occurs with the M.2 A+E Key Single 2.5g adaptor

Vohrt, 2025-03-09:
I have confirmed that the latest 6.8 kernel fixes this problem. The only requirement is that you cannot install the deb package in this repository or use the .sh file from the realtek official website, but need to use the official r8125 package

1. ethtool -K eth0 tx off sg off tso off where eth0 it's the address of your hardware NIC ( not the virtual bridge ).
2. disable iommu with bootup: add "intel_iommu=off“ in /etc/default/grub，Here is the detail link :https://pve.proxmox.com/wiki/Roadmap#Proxmox_VE_8.2
> intel_iommu defaulted to on in kernels 5.15 based kernels up to pve-kernel-5.15.35-3-pve with version 5.15.35-6. An enabled IOMMU can cause problems with older hardware - issues were reported with, for example, HP DL380 g8 servers, and Dell R610 servers (both older than 10 years).
The issue can be fixed by explicitly disabling intel_iommu on the kernel commandline (intel_iommu=off) following the reference documentation - https://pve.proxmox.com/pve-docs/chapter-sysadmin.html#sysboot_edit_kernel_cmdline
As the setting has been reverted in newer pve-kernel-5.15 packages the issue is now mostly relevant during installation from the ISO.
3. delete all r8125 package you installed before. You can use "apt remove realtek-r8125-dkms" to uninstall the deb in this repository. After that install the official one, just use "apt install r8125-dkms"
4. reboot and get into ther kernel 6.8, i found the network works well.

I don't know which specific operations worked, and these settings should cause the network card performance to decline, but at least it can run normally and stably now

closed 2025-03-09 by Vohrt