awesometic/realtek-r8125-dkms #56: manjaro 6.9.1 内核

kuailecs54, 2024-05-15:
manjaro 6.9.1 无法使用 安装脚本 6.8.9内核 可使用

kuailecs54, 2024-05-15:
请教一下 我需要修改什么来 在6.9.1 内核中使用

awesometic, 2024-05-16:
Hello,

Could you paste an error message when you're installing the package?

LsnmxNB, 2024-05-16:
https://www.realtek.com/Download/List?cate_id=584

参考，6.8可以

stefanh-it, 2024-05-26:
Please see all of the issues when trying to build this driver on kernel version 6.9.1 both as DKMS and directly from the upstream

output of dkms build log file: 

```
DKMS make.log for r8125-9.013.02 for kernel 6.8.9-zen1-1-zen (x86_64)
So 26 Mai 2024 23:14:32 CEST
/bin/sh: line 1: VER: command not found
make -C src/ KVER=6.8.9-zen1-1-zen BASEDIR=/lib/modules/6.9.2-zen1-1-zen modules
make[1]: Entering directory '/var/lib/dkms/r8125/9.013.02/build/src'
find: ‘/lib/modules/6.9.2-zen1-1-zen/kernel/drivers/net/ethernet’: No such file or directory
find: ‘/lib/modules/6.9.2-zen1-1-zen/kernel/drivers/net’: No such file or directory
make -C /lib/modules/6.9.2-zen1-1-zen/build M=/var/lib/dkms/r8125/9.013.02/build/src modules
make[2]: Entering directory '/var/lib/dkms/r8125/9.013.02/build/src'
make[2]: *** /lib/modules/6.9.2-zen1-1-zen/build: No such file or directory.  Stop.
make[2]: Leaving directory '/var/lib/dkms/r8125/9.013.02/build/src'
make[1]: *** [Makefile:188: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/r8125/9.013.02/build/src'
make: *** [Makefile:42: modules] Error 2
```

Install upstream source / Realtek website, I get this interesting result: 

```
/home/st/Documents/RealTek-Driver/r8125-9.013.02/src/r8125_n.c:7682:20: error: initialization of ‘int (*)(struct net_device *, struct ethtool_keee *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_eee *)’ [-Wincompatible-pointer-types]
 7682 |         .get_eee = rtl_ethtool_get_eee,
      |                    ^~~~~~~~~~~~~~~~~~~
/home/st/Documents/RealTek-Driver/r8125-9.013.02/src/r8125_n.c:7682:20: note: (near initialization for ‘rtl8125_ethtool_ops.get_eee’)
/home/st/Documents/RealTek-Driver/r8125-9.013.02/src/r8125_n.c:7683:20: error: initialization of ‘int (*)(struct net_device *, struct ethtool_keee *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_eee *)’ [-Wincompatible-pointer-types]
 7683 |         .set_eee = rtl_ethtool_set_eee,
      |                    ^~~~~~~~~~~~~~~~~~~
/home/st/Documents/RealTek-Driver/r8125-9.013.02/src/r8125_n.c:7683:20: note: (near initialization for ‘rtl8125_ethtool_ops.set_eee’)
make[4]: *** [scripts/Makefile.build:244: /home/st/Documents/RealTek-Driver/r8125-9.013.02/src/r8125_n.o] Error 1
make[3]: *** [/usr/lib/modules/6.9.2-zen1-1-zen/build/Makefile:1917: /home/st/Documents/RealTek-Driver/r8125-9.013.02/src] Error 2
make[2]: *** [Makefile:240: __sub-make] Error 2
make[1]: *** [Makefile:188: modules] Error 2
make: *** [Makefile:41: modules] Error 2

```

awesometic, 2024-05-30:
In kernel 6.9, there is a new network structure `ethtool_keee` that wraps `ethtool_eee`, which the current source uses.

- https://lore.kernel.org/all/a044621e-07f3-4387-9573-015f255db895@gmail.com/

Applying this makes things be changed to use `keee` to get `eee` properties.

So maybe we need to edit all of the parts related to that, not just change the variable/structure name but some simple logic to get the `eee`'s properties where it needs. 🤔 

For the best stability, I think we should wait for the official version by Realtek that supports kernel 6.9 or above.

stefanh-it, 2024-07-06:
Temp fix: 
There is a patch in the AUR comments which I am currenlty using until upstream is fixed
https://aur.archlinux.org/packages/r8125-dkms

Karlson2k, 2024-07-16:
> Temp fix: There is a patch in the AUR comments which I am currenlty using until upstream is fixed https://aur.archlinux.org/packages/r8125-dkms

It does not look like a fully correct patch.
This patch tries to convert parameters to u32 value, which is not enough for >1 Gbps speeds.

awesometic, 2024-07-17:
I can apply that patch to our project and test the performance between two machines supports 2.5 GbE.

I think it isn't harmful to the machines having kernel 6.8 or below, since the patch only affects if the detected kernel is 6.9 or up. But it can prevent from the machine couldn't use the Internet because of the build failure, so I think it is worth to apply while Realtek publishes the new version with supporting kernel 6.9.

So, probably, the new version will be uploaded soon.

Karlson2k, 2024-07-17:
@awesometic You may check Realtek implementation for the new kernel in the r8101 driver.
It has proper support for kernels >=6.9

awesometic, 2024-07-23:
A little late, I looked into how Realtek handles kernel 6.9 in their r8101 driver and can see that they implement `rtl_ethtool_get_eee`, `rtl_ethtool_set_eee` functions with link mode helpers like `linkmode_to_mii_eee_cap1_t` to use `ethtool_keee` rather than u32 values.

But in 2.5 GbE, as far as I investigate, it cannot be used since that function doesn't support 2.5 GbE link mode.
So I need to find another function that does a similar thing with that.
In addition, the corresponding functions (`rtl_ethtool_...`) in the r8125 source codes have more complex implementations. So maybe I can ruin the function works, so, I think it would be better to use u32 converters as **devome** did in AUR. It may reduce the performance as @Karlson2k said, but it works and makes our DKMS module successfully built on the new system. 🤔 

Let me use the u32 converters, and let's wait for Realtek's patch. Or, anybody can make PR for us. 😄

awesometic, 2024-11-16:
Hello,

According to the Realtek official download page, this issue is fixed in the latest driver version 9.014.01. So I will close the related issues 😄

closed 2024-11-16 by awesometic