awesometic/realtek-r8125-dkms #59: Data corruption on 9.013.02 ?

ewaldc, 2024-09-26:
I am in the process of rewriting the r8125 driver starting from your latest version 9.013.02.  The goal is to bring it closer to the mainline r8169 driver, reduce it's size, increase the performance/reduce system load while keeping the additional features such as RSS, PTP etc.
While testing the 3 versions (9.013.02, r8125 new, r8169 6.8 mainline backported to 6.1 to fix several data corruptions) side-by-side, it was noticed that a few of the 9.013.02 tests failed. 
The test platform is a Rockchip rk3588 system with dual 8125B running Ubuntu Jammy 24.04.  All drivers were compiled using gcc14.2.0 cross compiler for arm64 and used MTU 1500/9000 for each round of testing.  For  9.013.02, all was left untouched including the Makefile.  Kernel is 6.1.

On one of the more challenging tests (a 500 MB zstd compressed archive containing many binary files) :
```
wget "http://ports.ubuntu.com/pool/main/l/linux-firmware/linux-firmware_20240913.gita34e7a5f-0ubuntu2_arm64.deb"
```
both the mainline r8169 and the new r8125 driver have the (correct) sha512 checksum (`cksum -a sha512`) of `3956d1faa6a5aebf624a42d39883594b70562d3919545cb4c18c44842df16eceddbc1a79867482c88310f63cc5033b13e1590472a4e11695d5a75ee23683905f` but the original 9.013.02 driver is giving different checksum results at each download.  

Is there any chance to execute this test ?   If not reproducable, I need to find the mistake from my side or look into external factors like gcc version, kernel version/corruption, system HW etc.  Alternatively, if confirmed, it would be good for users to know...

Thanks in advance.

ewaldc, 2024-12-22:
I managed to find several root causes for these corruptions.

- with lots of small packets, frags could contain the wrong value.  This also hit the [mainline r8169 driver ](https://lore.kernel.org/linux-cve-announce/2024061949-CVE-2024-38586-70d6@gregkh/) 
- memory barriers missing and/or incorrectly positioned + some additional READ_ONCE/WRITE_ONCE required, causing issues with (fast) 6+ core CPU's like RK3588.
I posted a rewrite of the driver [here](https://github.com/ewaldc/realtek-r8125-dkms), it's now much closer to the mainline r8169 version, about half the size and consumes less memory and CPU resources.

Pix13, 2025-04-16:
Hi @ewaldc do you plan to make a MR or publish a fork of your driver ? I have some 3588 with heavy use and your description looks promising :)

ewaldc, 2025-04-18:
Hello @Pix13, you can find it [here](https://github.com/ewaldc/realtek-r8125-dkms).  Just published my current working version, hope I got all the right sources. It's quite a major rewrite.  Haven't tested the dkms part because it's running on cluster of rk3588 systems which are updated only once a year, nor did I have time to look into v9.15.  I made progress on the PTP part, but it's not working yet in a reliable fashion.