awesometic/realtek-r8125-dkms #60: DKMS Build Fails under kernel version 6.10.10-zen1-1-zen

SipSup3314, 2024-09-30:
```sh
oil@oliman-arch | ~ $ sudo dkms install r8125/9.013.02
Sign command: /usr/lib/modules/6.10.10-zen1-1-zen/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j20 KERNELRELEASE=6.10.10-zen1-1-zen -C /usr/lib/modules/6.10.10-zen1-1-zen/build M=/var/lib/dkms/r8125/9.013.02/build modules...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.10.10-zen1-1-zen (x86_64)
Consult /var/lib/dkms/r8125/9.013.02/build/make.log for more information.
```
The make.log file shows the following
```
DKMS make.log for r8125-9.013.02 for kernel 6.10.10-zen1-1-zen (x86_64)
Mon Sep 30 05:22:10 PM EDT 2024
make: Entering directory '/usr/lib/modules/6.10.10-zen1-1-zen/build'
  CC [M]  /var/lib/dkms/r8125/9.013.02/build/r8125_n.o
  CC [M]  /var/lib/dkms/r8125/9.013.02/build/rtl_eeprom.o
  CC [M]  /var/lib/dkms/r8125/9.013.02/build/rtltool.o
/var/lib/dkms/r8125/9.013.02/build/r8125_n.c:7682:20: error: initialization of ‘int (*)(struct net_device *, struct ethtool_keee *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_eee *)’ [-Wincompatible-pointer-types]
 7682 |         .get_eee = rtl_ethtool_get_eee,
      |                    ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/r8125/9.013.02/build/r8125_n.c:7682:20: note: (near initialization for ‘rtl8125_ethtool_ops.get_eee’)
/var/lib/dkms/r8125/9.013.02/build/r8125_n.c:7683:20: error: initialization of ‘int (*)(struct net_device *, struct ethtool_keee *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_eee *)’ [-Wincompatible-pointer-types]
 7683 |         .set_eee = rtl_ethtool_set_eee,
      |                    ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/r8125/9.013.02/build/r8125_n.c:7683:20: note: (near initialization for ‘rtl8125_ethtool_ops.set_eee’)
make[2]: *** [scripts/Makefile.build:244: /var/lib/dkms/r8125/9.013.02/build/r8125_n.o] Error 1
make[1]: *** [/usr/lib/modules/6.10.10-zen1-1-zen/build/Makefile:1932: /var/lib/dkms/r8125/9.013.02/build] Error 2
make: *** [Makefile:240: __sub-make] Error 2
make: Leaving directory '/usr/lib/modules/6.10.10-zen1-1-zen/build'
```
I am not too experienced in C so I'm just gonna leave this to the pros.

dct-infra, 2024-10-02:
Hello,
I'm Facing the exact same errors here.

awesometic, 2024-10-02:
Hello,

It is the same issue as Issue #56 , please take a look at this one for more information.

dct-infra, 2024-10-02:
@awesometic Thank you for your reply.
Loded and now the "default" r8169 keeps loading along with the r8125 driver on one machine and not the other...
```
02:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
	Subsystem: Realtek Semiconductor Co., Ltd. Device 0123
	Kernel driver in use: r8169
	Kernel modules: r8169, r8125
```
03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
	Subsystem: Realtek Semiconductor Co., Ltd. Device 0123
	Kernel driver in use: r8125
	Kernel modules: r8125

What a headache :)
I'll get back to my old 1G nic for now.

awesometic, 2024-11-16:
Hello,

According to the Realtek official download page, this issue is fixed in the latest driver version 9.014.01. So I will close the related issues 😄

closed 2024-11-16 by awesometic