awesometic/realtek-r8125-dkms #61: Request to create realtek-r8126-dkms

wjz304, 2024-10-03:
https://www.realtek.com/Download/List?cate_id=584#:~:text=5G%20Ethernet%20LINUX%20driver%20r8126%20for%20kernel%20up%20to%206.4

wjz304, 2024-10-04:
Yes, I can test it, I have 8126 device, (https://www.wisdpi.com/products/wisdpi-pcie-5g-base-t-ethernet-adapter?pr_prod_strat=e5_desc&pr_rec_id=6fba14b9e&pr_rec_pid=8586582524158&pr_ref_pid=8570406404350&pr_seq=uniform)

The driver has been tested OK, but because 8125 and 8152 use your warehouse, 8126 wants to use your warehouse as well.

awesometic, 2024-10-04:
Okay, then I will make it and let you know the results 😄

awesometic, 2024-10-06:
Hello,

Please check this release and can you share the results?
- https://github.com/awesometic/realtek-r8126-dkms/releases/tag/10.013.00-1

wjz304, 2024-10-06:
```
root@N100-4:~# dpkg -i realtek-r8126-dkms_10.013.00-1_amd64.deb
Selecting previously unselected package realtek-r8126-dkms.
(Reading database ... 94584 files and directories currently installed.)
Preparing to unpack realtek-r8126-dkms_10.013.00-1_amd64.deb ...
Unpacking realtek-r8126-dkms (10.013.00-1) ...
Setting up realtek-r8126-dkms (10.013.00-1) ...
Loading new realtek-r8126-10.013.00 DKMS files...
Deprecated feature: REMAKE_INITRD (/usr/src/realtek-r8126-10.013.00/dkms.conf)
Building for 6.8.4-2-pve
Building for architecture amd64
Building initial module for 6.8.4-2-pve
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/realtek-r8126/10.013.00/source/dkms.conf)
Done.
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/realtek-r8126/10.013.00/source/dkms.conf)
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/realtek-r8126/10.013.00/source/dkms.conf)

r8126.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.8.4-2-pve/updates/dkms/
depmod...

root@N100-4:~# modinfo r8126
filename:       /lib/modules/6.8.4-2-pve/updates/dkms/r8126.ko
version:        10.013.00-NAPI
license:        GPL
description:    Realtek r8126 Ethernet controller driver
author:         Realtek and the Linux r8126 crew <netdev@vger.kernel.org>
srcversion:     DC01E378AB591B461A95B9D
alias:          pci:v000010ECd00005000sv*sd*bc*sc*i*
alias:          pci:v000010ECd00008126sv*sd*bc*sc*i*
depends:        
retpoline:      Y
name:           r8126
vermagic:       6.8.4-2-pve SMP preempt mod_unload modversions 
sig_id:         PKCS#7
signer:         DKMS module signing key
sig_key:        ...
sig_hashalgo:   sha512
signature:      ...
parm:           speed_mode:force phy operation. Deprecated by ethtool (8). (uint)
parm:           duplex_mode:force phy operation. Deprecated by ethtool (8). (uint)
parm:           autoneg_mode:force phy operation. Deprecated by ethtool (8). (uint)
parm:           aspm:Enable ASPM. (int)
parm:           s5wol:Enable Shutdown Wake On Lan. (int)
parm:           s5_keep_curr_mac:Enable Shutdown Keep Current MAC Address. (int)
parm:           use_dac:Enable PCI DAC. Unsafe on 32 bit PCI slot. (int)
parm:           timer_count:Timer Interrupt Interval. (int)
parm:           eee_enable:Enable Energy Efficient Ethernet. (int)
parm:           hwoptimize:Enable HW optimization function. (ulong)
parm:           s0_magic_packet:Enable S0 Magic Packet. (int)
parm:           tx_no_close_enable:Enable TX No Close. (int)
parm:           enable_ptp_master_mode:Enable PTP Master Mode. (int)
parm:           disable_wol_support:Disable PM support. (int)
parm:           enable_double_vlan:Enable Double VLAN. (int)
parm:           debug:Debug verbosity level (0=none, ..., 16=all) (int)

root@N100-4:~# ls /sys/class/net/*/device/driver -ld
lrwxrwxrwx 1 root root 0 Oct  5 16:42 /sys/class/net/enp3s0/device/driver -> ../../../../bus/pci/drivers/r8126
lrwxrwxrwx 1 root root 0 Oct  5 16:42 /sys/class/net/enp4s0/device/driver -> ../../../../bus/pci/drivers/igc
lrwxrwxrwx 1 root root 0 Oct  5 16:42 /sys/class/net/enp5s0/device/driver -> ../../../../bus/pci/drivers/igc
lrwxrwxrwx 1 root root 0 Oct  5 16:42 /sys/class/net/enp6s0/device/driver -> ../../../../bus/pci/drivers/igc
lrwxrwxrwx 1 root root 0 Oct  5 16:42 /sys/class/net/enp7s0/device/driver -> ../../../../bus/pci/drivers/igc

root@N100-4:~# ethtool enp3s0
Settings for enp3s0:
        Supported ports: [ TP ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                2500baseT/Full
                                5000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                2500baseT/Full
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Speed: Unknown!
        Duplex: Unknown! (255)
        Auto-negotiation: on
        Port: Twisted Pair
        PHYAD: 0
        Transceiver: internal
        MDI-X: on
        Supports Wake-on: pumbg
        Wake-on: g
        Current message level: 0x00000033 (51)
                               drv probe ifdown ifup
        Link detected: no
```

wjz304, 2024-10-07:
I haven't found any problems in the test so far. Due to the hard disk limitation, I haven't tested the actual read and write speeds.

Let's close this issue first.
If there are any questions, I will create a new issue at https://github.com/awesometic/realtek-r8126-dkms

awesometic, 2024-10-07:
Thank you for sharing the results. I will close this issue then 😄

closed 2024-10-07 by awesometic