awesometic/realtek-r8125-dkms #62: Deprecated feature: REMAKE_INITRD

Br33ce, 2024-10-04:
After the latest Kernel Update to 6.1.0-26-amd64 I noticed something in the log:

`dkms: running auto installation service for kernel 6.1.0-26-amd64.
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/realtek-r8125/9.013.02/source/dkms.conf)
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/realtek-r8125/9.013.02/source/dkms.conf)
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/realtek-r8125/9.013.02/source/dkms.conf)
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/realtek-r8125/9.013.02/source/dkms.conf)
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/realtek-r8125/9.013.02/source/dkms.conf)
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/realtek-r8125/9.013.02/source/dkms.conf)
Deprecated feature: REMAKE_INITRD (/etc/dkms/framework.conf)
Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/realtek-r8125/9.013.02/source/dkms.conf)`

Is that something that should "worry" me because the deprecated feature might stop working in the "near" future or can it be ignored?

If necessary I could provide the whole log. Thank you for your work 😊

Br33ce, 2024-10-07:
Here is the complete log @awesometic 😊
https://pastebin.com/65CCXFzF

So far I can't see anything wrong. The required reboot, I had a [problem](https://github.com/awesometic/realtek-r8125-dkms/issues/51) in the past, worked.
Nothing to worry about then? 😊

awesometic, 2024-10-07:
If we set `REMAKE_INITRD` to the default value of `REMAKE_INITRD`, there will be no problem since it will just ignore the flag. But if it doesn't, we need to do something before actually removed.

But it seems like it is already removed from the dkms binary and the default behavior is "no". So I think we just can remove that flag from `dkms.conf`

If you don't mind, how about testing without `REMAKE_INITRD` and make a PR? 😄

Br33ce, 2024-10-07:
To be honest I don't really know how to remove the "flag" and how to test it (on my live system 😱).

awesometic, 2024-10-07:
Ah okay, doing a test on a live system can lead to unwanted consequences 🙈 Then I will remove that deprecated flag and update the package :)

Br33ce, 2024-10-07:
Thank you 😊 should I then simply install the package again or will it update as other packages (apt-get upgrade..)?

awesometic, 2024-10-08:
Use `apt` to update the package 😄 But there will be no feature-side changes

awesometic, 2024-11-16:
This deprecated flag is removed from the latest Debian package version 9.014.01-1. Thanks. 😄

closed 2024-11-16 by awesometic