awesometic/realtek-r8125-dkms #63: Cannot increase the speed above 100Mbps

Incantatoric, 2024-10-21:
Hi, so I followed the instruction but I cannot increase the speed. If I use ethtool to force increasing the speed, it just disconnects or sometimes gives me error message that it is invalid command. I am dual booting on Windows on a separate drive and there I get higher speed. Below is the information:

# Kernel Version
uname -r
6.8.0-47-generic

# Installed Kernels
dpkg --list | grep linux-image
ii  linux-image-6.8.0-47-generic                  6.8.0-47.47                              amd64        Signed kernel image generic
ii  linux-image-generic-hwe-24.04                 6.8.0-47.47                              amd64        Generic Linux kernel image

# Driver in Use
lspci -k | grep -A 3 Ethernet
0e:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
        Subsystem: ASRock Incorporation RTL8125 2.5GbE Controller
        Kernel driver in use: r8125
        Kernel modules: r8169, r8125

# Network Device Info
sudo lshw -C network
  *-network                 
       description: Ethernet interface
       product: RTL8125 2.5GbE Controller
       vendor: Realtek Semiconductor Co., Ltd.
       logical name: enp14s0
       size: 100Mbit/s
       capacity: 1Gbit/s
       driver: r8125
       driverversion: 9.013.02-NAPI
       speed: 100Mbit/s

# ethtool Output
sudo ethtool enp14s0
Speed: 100Mb/s
Duplex: Full
Auto-negotiation: on

# Check Loaded Modules
lsmod | grep r8125
r8125                 286720  0

# dmesg Output (Errors/Warnings)
sudo dmesg | grep r8125
[ 4.004697] r8125: loading out-of-tree module taints kernel.
[ 4.004706] r8125: module verification failed: signature and/or required key missing - tainting kernel
[ 4.093760] r8125 0000:0e:00.0 enp14s0: renamed from eth0
[ 270.914457] r8125 0000:0e:00.0 enp14s0: Transmit timeout reset Device!
[ 270.919438] r8125 0000:0e:00.0 enp14s0: Device reseting!

# Attempt to Set 1000 Mbps (Error message or disconnection)
sudo ethtool -s enp14s0 speed 1000 duplex full autoneg on

# Network Manager Logs
journalctl -u NetworkManager

# Network Interface Info
ip link show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
3: enp14s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode DEFAULT group default qlen 1000

I have no idea why this is causing an error. I have tried everything that I could try. I even bought a usb-ethernet adapter but it did not change anything. If needed, I can buy an intel-based network card but I am not even sure if that will fix the issue... I will provide with you any detail that is necessary. Thank you in advance.

awesometic, 2024-10-29:
Hello,

It seems like the r8125 module is not loaded properly. How did you install the package? Using the deb file or PPA?

Incantatoric, 2024-10-30:
> Hello,
> 
> It seems like the r8125 module is not loaded properly. How did you install the package? Using the deb file or PPA?

Hi, thank you for your reply! I think I did it with deb. Should I uninstall it and then try with PPA?

awesometic, 2024-10-30:
Installing it with the deb file should work.

Uninstall it and reinstall the deb, or PPA if you want to, and please check this contents after installation.
https://github.com/awesometic/realtek-r8125-dkms?tab=readme-ov-file#verify-the-module-is-loaded-successfully

If still it doesn't work, please let me know.

Incantatoric, 2024-11-28:
> Installing it with the deb file should work.
>
> Uninstall it and reinstall the deb, or PPA if you want to, and please check this contents after installation. [...]

Hi, I tried with the latest .deb file but the result is almost the same. This latest version had no error message when installing unlike the previous one, but the symptoms are almost the same:

# sudo lshw -C network
  *-network
       description: Ethernet interface
       product: RTL8125 2.5GbE Controller
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0000:0e:00.0
       logical name: enp14s0
       version: 05
       serial: 9c:6b:00:63:b7:4f
       size: 100Mbit/s
       capacity: 1Gbit/s
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress msix vpd bus_master cap_list ethernet physical tp 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=r8125 driverversion=9.014.01-NAPI duplex=full ip=192.168.45.77 latency=0 link=yes multicast=yes port=twisted pair speed=100Mbit/s
       resources: irq:38 ioport:e000(size=256) memory:f6a00000-f6a0ffff memory:f6a10000-f6a13fff

# sudo ethtool enp14s0
Settings for enp14s0:
	Supported ports: [ TP ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseT/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseT/Full
	Advertised pause frame use: Symmetric
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	Link partner advertised pause frame use: Transmit-only
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 100Mb/s
	Duplex: Full
	Auto-negotiation: on
	Port: Twisted Pair
	PHYAD: 0
	Transceiver: internal
	MDI-X: on
	Supports Wake-on: pumbg
	Wake-on: g
        Current message level: 0x00000033 (51)
                               drv probe ifdown ifup
	Link detected: yes

# lsmod | grep r8125
r8125                 303104  0
# lsmod | grep r8169

# sudo dmesg | grep r8125
[    4.872519] r8125: loading out-of-tree module taints kernel.
[    4.872530] r8125: module verification failed: signature and/or required key missing - tainting kernel
[    4.874632] r8125 Ethernet controller driver 9.014.01-NAPI loaded
[    5.264612] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[    5.266635] r8125  Copyright (C) 2024 Realtek NIC software team <nicfae@realtek.com>
[    5.980030] r8125 0000:0e:00.0 enp14s0: renamed from eth0
[   21.235359] r8125: enp14s0: link up
[  110.718667] r8125: enp14s0: link down
[  124.037049] r8125: enp14s0: link up
[  328.916948] r8125: enp14s0: link down
[  328.927715] workqueue: rtl8125_linkchg_task [r8125] hogged CPU for >10000us 4 times, consider switching to WQ_UNBOUND
[  342.185667] r8125: enp14s0: link up
[  406.382325] r8125: enp14s0: link down
[  419.732876] r8125: enp14s0: link up
[  419.749336] workqueue: rtl8125_linkchg_task [r8125] hogged CPU for >10000us 8 times, consider switching to WQ_UNBOUND
[  419.754211] r8125: enp14s0: link down
[  433.025386] r8125: enp14s0: link up
[  433.046774] r8125: enp14s0: link down
[  448.986398] r8125: enp14s0: link up
[  449.007658] r8125: enp14s0: link down
[  462.215256] r8125: enp14s0: link up

I tried setting the change 1000 and it is now appropriately shown at system settings network, but no matter what I try with link negotiation and manual speed, I am stuck at 100MB...