awesometic/realtek-r8125-dkms #66: The latest version suffers from severe disconnection issues, and the network speed doesn't exceed 100 Mbps.

jcfun, 2024-11-18:
I tested the latest driver on two machines: one running Arch Linux (6.11.8-zen1-2-zen) and the other running PVE (6.8.12-4-pve). I found that this driver seems to have significant problems. Both machines experienced severe disconnections and network speeds below 100 Mbps when tested with iperf3 (the experience in daily use was even worse). These issues disappeared after I reverted the driver to version 9.013.02-2, so I suspect there is a severe bug in version 9.014.01-1.

jcfun, 2024-11-18:
[image]
[image]
[image]
The above is the case when only the archlinux host installs the 9.014.01-1 version of the driver. If two drivers are installed at the same time, the iperf3 test will be much worse.

hireworksltd, 2024-11-21:
Interesting! not seeing the same issue here on 6.12 with 9.014.01-1. Have run speedtest for 24h and getting 1Gbps up/down. (haven't tried 2.5Gbit yet).

Cheers,
Jon

awesometic, 2024-11-22:
> If two drivers are installed at the same time, the iperf3 test will be much worse.

Did you mean r8169 and r8125 at the same time?

jcfun, 2024-11-22:
> > 如果同时安装两个驱动的话，iperf3测试的结果会差很多。
> 
> 您指的是 r8169 和 r8125 吗？

No, I mean that the performance will be worse when the r8125 driver is installed on both hosts and iperf3 is tested.

jcfun, 2024-11-22:
> > If two drivers are installed at the same time, the iperf3 test will be much worse.
> 
> Did you mean r8169 and r8125 at the same time?

No, I mean that the performance will be worse when the r8125 driver is installed on both hosts and iperf3 is tested.

ewaldc, 2024-12-22:
There are several issues with 9.014.01-1, but never saw disconnection troubles or dismal performance, nor issues between systems running the r8169 mainline drivers and systems running 9.014.01-1.  I achieve between 2.35Gb/s and 2.48Gb/s using iperf3 on both 2.5Gb and 10Gb switches in both directions, depending on the MTU size (1500 or 9000).   There is not much difference between 9.014.01-1 and 9.013.02-2, but the `RTL_R8(tp, IMR0_8125);` that was added in `rtl8125_rx_interrupt` could be a possible cause.

On the other side, I struggled with hangs and data corruption and ended up [rewriting the driver](https://github.com/ewaldc/realtek-r8125-dkms).   You are free to give it a try, but I have not yet tested it on kernels above 6.1 and only tested on Orange Pi 5+ system (dual 8125B).
I assume you installed the correct PVE headers ? e.g. `apt install -y dkms build-essential pve-headers-$(uname -r)`