awesometic/realtek-r8125-dkms #67: ptp硬件对时问题

yanghappyfrog, 2025-01-15:
在香橙派上使用ptp对时,会报如下错误:
orangepi@orangepi5plus:~/test$ sudo ptp4l -m -H -s -i enP4p65s0
ptp4l[889.438]: selected /dev/ptp1 as PTP clock
ptp4l[889.440]: port 1: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[889.440]: port 0: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[896.371]: selected local clock c0742b.fffe.fd98b1 as best master
ptp4l[900.918]: port 1: new foreign master 2cea7f.fffe.eafbdc-1
ptp4l[904.342]: selected local clock c0742b.fffe.fd98b1 as best master
ptp4l[904.918]: selected best master clock 2cea7f.fffe.eafbdc
ptp4l[904.918]: running in a temporal vortex
ptp4l[904.918]: port 1: LISTENING to UNCALIBRATED on RS_SLAVE
ptp4l[905.471]: timed out while polling for tx timestamp
ptp4l[905.471]: increasing tx_timestamp_timeout may correct this issue, but it is likely caused by a driver bug
ptp4l[905.471]: port 1: send delay request failed
ptp4l[905.471]: port 1: UNCALIBRATED to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
ptp4l[921.474]: port 1: FAULTY to LISTENING on INIT_COMPLETE
ptp4l[922.920]: port 1: new foreign master 2cea7f.fffe.eafbdc-1
ptp4l[926.921]: selected best master clock 2cea7f.fffe.eafbdc
ptp4l[926.921]: running in a temporal vortex
ptp4l[926.921]: port 1: LISTENING to UNCALIBRATED on RS_SLAVE
ptp4l[928.839]: timed out while polling for tx timestamp
ptp4l[928.839]: increasing tx_timestamp_timeout may correct this issue, but it is likely caused by a driver bug
ptp4l[928.839]: port 1: send delay request failed
ptp4l[928.839]: port 1: UNCALIBRATED to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
驱动版本:realtek-r8125-dkms-9.014.01-1

bpx69, 2025-03-04:
I have been playing with liuxptp 4.4 and r8125 driver in the last days.
The driver does not work for certain linuxptp configurations and it works for other. 

For example it does not work with BMCA set to noop and clientOnly 1 from automotive-slave configuration. It does not work with no configuration specified.

It worked for example using this gPTP config:
`[global]
#
# Default Data Set
#
twoStepFlag		1
clientOnly		0
socket_priority		0
priority1		128
priority2		128
domainNumber		0
#utc_offset		37
clockClass		248
clockAccuracy		0xFE
offsetScaledLogVariance	0xFFFF
free_running		0
freq_est_interval	1
dscp_event		0
dscp_general		0
dataset_comparison	ieee1588
G.8275.defaultDS.localPriority	128
maxStepsRemoved		255
#
# Port Data Set
#
logAnnounceInterval	1
logSyncInterval		0
operLogSyncInterval	0
logMinDelayReqInterval	0
logMinPdelayReqInterval	0
operLogPdelayReqInterval	0
announceReceiptTimeout	3
syncReceiptTimeout	0
delay_response_timeout	0
delayAsymmetry		0
fault_reset_interval	4
neighborPropDelayThresh	20000000
serverOnly		0
G.8275.portDS.localPriority	128
allowedLostResponses	3
asCapable		auto
BMCA			ptp
inhibit_announce	0
inhibit_delay_req	0
ignore_source_id	0
power_profile.2011.grandmasterTimeInaccuracy	-1
power_profile.2011.networkTimeInaccuracy	-1
power_profile.2017.totalTimeInaccuracy		-1
power_profile.grandmasterID			0
power_profile.version				none
ptp_minor_version	0
spp			-1
active_key_id		0
#
# Run time options
#
assume_two_step		1
logging_level		6
path_trace_enabled	0
follow_up_info		0
hybrid_e2e		0
inhibit_multicast_service	0
net_sync_monitor	0
tc_spanning_tree	0
tx_timestamp_timeout	10
unicast_listen		0
unicast_master_table	0
unicast_req_duration	3600
use_syslog		1
verbose			0
summary_interval	0
kernel_leap		1
check_fup_sync		0
clock_class_threshold	248
#
# Servo Options
#
pi_proportional_const	0.0
pi_integral_const	0.0
pi_proportional_scale	0.0
pi_proportional_exponent	-0.3
pi_proportional_norm_max	0.7
pi_integral_scale	0.0
pi_integral_exponent	0.4
pi_integral_norm_max	0.3
step_threshold		0.0
first_step_threshold	0.00002
max_frequency		900000000
clock_servo		pi
sanity_freq_limit	200000000
refclock_sock_address	/var/run/refclock.ptp.sock
ntpshm_segment		0
msg_interval_request	0
servo_num_offset_values	10
servo_offset_threshold	0
write_phase_mode	0
#
# Transport options
#
cmlds.client_address	/var/run/cmlds_client
cmlds.domainNumber	0
cmlds.majorSdoId	2
cmlds.port		0
cmlds.server_address	/var/run/cmlds_server
transportSpecific	0x0
ptp_dst_ipv4		224.0.1.129
p2p_dst_ipv4		224.0.0.107
ptp_dst_ipv6		FF0E:0:0:0:0:0:0:181
p2p_dst_ipv6		FF02:0:0:0:0:0:0:6B
ptp_dst_mac		01:1B:19:00:00:00
p2p_dst_mac		01:80:C2:00:00:0E
udp_ttl			1
udp6_scope		0x0E
uds_address		/var/run/ptp4l
uds_file_mode		0660
uds_ro_address		/var/run/ptp4lro
uds_ro_file_mode	0666
#
# Default interface options
#
clock_type		OC
network_transport	UDPv4
delay_mechanism		E2E
time_stamping		hardware
tsproc_mode		filter
delay_filter		moving_median
delay_filter_length	10
egressLatency		0
ingressLatency		0
boundary_clock_jbod	0
phc_index		-1
#
# Clock description
#
productDescription	;;
revisionData		;;
manufacturerIdentity	00:00:00
userDescription		;
timeSource		0xA0`

bpx69, 2025-03-04:
As far as I could trace the driver with a non-working config, the chip sets the PTP_ISR_TER flag in PTP_ISR_8125 register (which I think means timestamping error) instead of PTP_ISR_TOK (which IMO means timestamping went ok). The driver then does not put the timestamped message back on the MSG_ERRQUEUE and ptp4l reports an error it cannot read the timestamp back.