awesometic/realtek-r8125-dkms #70: Does r8125 driver support other realtek models?

narcisgarcia, 2025-03-05:
I have a scenario with RTL8168 and RTL8125 devices.
Realtek 8168 is working with r8169 module.
What happens if I install r8125 module and blacklist r8169?
Will r8125 module manage all Realtek devices or will it leave me without other NICs networking?

Thank you.

awesometic, 2025-03-09:
Hello,

Please check the quotes part of this README section: https://github.com/awesometic/realtek-r8125-dkms?tab=readme-ov-file#verify-the-module-is-loaded-successfully

If you install r8125 module and blacklist r8169, only RTL8125 will work and RTL8168 won't. In the README blacklisting r8169 device is for preventing from loading RTL8125 by r8169 module. If you remove the RTL8125 firmware files, r8169 module will not load RTL8125 device but RTL8168, and installed r8125 module will load RTL8125 device.

You need to test this on your side enoughly before using this workaround for your daily working. If it doesn't work, please let me know to update README.md file 😄

narcisgarcia, 2025-03-09:
Well I need to use both RTL8125 and RTL8168 devices on same host.
I prefer to have RTL8125 ready for 2.5gbps, but I can't sacrifice the use of RTL8168 NIC.

If r8169 supports both RTL8168 and RTL8125 devices and r8125 is not compatible with RTL8168, then I must use r8169.
Hope r8169 fully supports RTL8125 at 1gbps at least.

awesometic, 2025-03-31:
Using r8169 with r8125 firmware, which is how r8169 module support various chips, may work with 2.5 gbps if I remember right. The built-in module and firmware depends on your kernel version.