awesometic/realtek-r8125-dkms #72: Issue with kernel 6.11.0.21 and realtek driver 9.015

jordicm10, 2025-03-28:
Hi All,

I have Ubuntu 24.04.02 LTS with kernel hwe and the realtek driver installed from the Launchpad ppa repository.

Today the kernel has been updated to 6.11.0.21 (from 6.11.0.19 I think) and after reboot the ethernet interface has disappeared (no internet connection).

More info:

```
uname -a
Linux 6.11.0-21-generic #21~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Feb 24 16:52:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

lspci -k
01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 0c)
	Subsystem: ASUSTeK Computer Inc. RTL8125 2.5GbE Controller
	Kernel driver in use: r8125
	Kernel modules: r8169, r8125
```

awesometic, 2025-03-29:
Hello,

How about rebuilding module using `dkms`?

```
sudo dkms install -m realtek-r8125/9.015.00 -k $(uname -r) --force
```

In general, it should build the module during in progress of updating kernel, but it seems like something interfered in your case. Can't sure, please share the result if you run this command.

jordicm10, 2025-03-30:
This is the result

```
$ sudo dkms install -m realtek-r8125/9.015.00 -k $(uname -r) --force
Module realtek-r8125-9.015.00 for kernel 6.11.0-21-generic (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

r8125.ko.zst:
 - Uninstallation
   - Deleting from: /lib/modules/6.11.0-21-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
depmod...

r8125.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.11.0-21-generic/updates/dkms/
depmod...
```
After a reboot the problem is the same.

Just for info, the ethernet interface is there:
```
$ ifconfig 
enp1s0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 48:21:0b:6f:89:ec  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 18 
```
but Ubuntu says there is no cable connected.

I would have suspected a HW failure, it failed just after a reboot with to use the last hwe kernel.

awesometic, 2025-03-31:
Can you share the result of the below commands?

```
$ dmesg | grep r8125
$ dmesg | grep -i realtek
$ lsmod | grep -E 'r8125|r8169'
```

jordicm10, 2025-03-31:
```
$ dmesg | grep r8125

$ dmesg | grep -i realtek

$ lsmod | grep -E 'r8125|r8169'
r8125                 307200  0
r8169                 118784  0
```

awesometic, 2025-03-31:
How about `dmesg | grep r8125` right after reboot system?

And it looks like the system loads two modules simultaneously, you can prevent from loading r8169 by blacklisting that module according to this guide, if you don't need r8169: https://github.com/awesometic/realtek-r8125-dkms?tab=readme-ov-file#verify-the-module-is-loaded-successfully

awesometic, 2025-04-03:
It seems like the hardware doesn't response. How about test it using live image like Ubuntu 24.04 live? Because it should work with the built-in r8169 driver, so you can check the hardware is corrupted or not.

barabeslo, 2025-04-03:
Hi! I have similar problems, after installing the r8125, the network card disappeared

`$ uname -a
`
`Linux compucter 6.11.0-21-generic #21-Ubuntu SMP PREEMPT_DYNAMIC Wed Feb 19 16:50:40 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux
`

`lspci -k`
`0b:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
        Subsystem: Micro-Star International Co., Ltd. [MSI] Device 7d75
        Kernel modules: r8125
`

There is no line

`Kernel driver in use: r8125`

jordicm10, 2025-04-03:
> It seems like the hardware doesn't response. How about test it using live image like Ubuntu 24.04 live? Because it should work with the built-in r8169 driver, so you can check the hardware is corrupted or not.

I have rebooted into kernel 6.11.0-19-generic and installed realtek driver 9.014, that worked without problems previously.
The ethernet interface is there but it does not detect cable connection, so I must conclude that there is some problem with the hw.

jordicm10, 2025-04-03:
I just realized that since the kernel and driver were update around one week ago, I have security events related to Intel Boot Guard and Linux Kernel Verification.

Could it be that the driver is not allowed to run by UEFI Secure Boot?

awesometic, 2025-04-04:
> I have rebooted into kernel 6.11.0-19-generic and installed realtek driver 9.014, that worked without problems previously. The ethernet interface is there but it does not detect cable connection, so [...]

Weird. How about 9.014.01 with the current kernel? If it works back, then 9.015 could be the culprit.

> Could it be that the driver is not allowed to run by UEFI Secure Boot?

If so, I think the driver wouldn't be loaded and shows another error message on dmesg. Do you enable secure boot?

jordicm10, 2025-04-04:
9.014 with kernel 6.11.0-21 does not work.

Secure boot is always enabled. I did a quick test disabling Secure boot with no luck.

This is the boot log with 9.014:
```
$ sudo dmesg | grep r8125
[sudo] contraseña para jordi: 
[   15.345017] r8125: loading out-of-tree module taints kernel.
[   15.356515] r8125 Ethernet controller driver 9.014.01-NAPI loaded
[   15.356571] r8125 0000:01:00.0: enabling device (0000 -> 0003)
[   15.645098] WARNING: CPU: 1 PID: 395 at /var/lib/dkms/realtek-r8125/9.014.01/build/src/r8125_n.c:8022 rtl8125_wait_phy_ups_resume+0xb8/0xd0 [r8125]
[   15.645111] Modules linked in: snd_seq_midi_event cmdlinepart snd_rawmidi processor_thermal_device_pci uvcvideo(+) spi_nor(+) spd5118 processor_thermal_device videobuf2_vmalloc iwlmvm(+) drm_buddy snd_seq processor_thermal_wt_hint uvc snd_seq_device mei_hdcp mtd mei_pxp videobuf2_memops btusb processor_thermal_rfim ttm intel_rapl_msr(+) mac80211 snd_timer videobuf2_v4l2 btrtl drm_display_helper processor_thermal_rapl libarc4 btintel asus_nb_wmi mfd_aaeon videodev i2c_i801 intel_rapl_common cec btbcm processor_thermal_wt_req snd asus_wmi i2c_mux intel_cstate(+) spi_intel_pci iwlwifi videobuf2_common btmtk processor_thermal_power_floor rc_core mei_me platform_profile r8125(O+) soundcore i2c_smbus wmi_bmof cfg80211 spi_intel mc processor_thermal_mbox bluetooth mei i2c_algo_bit int340x_thermal_zone intel_pmc_core igen6_edac intel_vsec pmt_telemetry intel_hid pmt_class sparse_keymap acpi_pad acpi_tad joydev input_leds mac_hid sch_fq_codel msr parport_pc ppdev lp parport efi_pstore nfnetlink dmi_sysfs ip_tables x_tables
[   15.645175] RIP: 0010:rtl8125_wait_phy_ups_resume+0xb8/0xd0 [r8125]
[   15.645208]  ? rtl8125_wait_phy_ups_resume+0xb8/0xd0 [r8125]
[   15.645228]  ? rtl8125_wait_phy_ups_resume+0xb8/0xd0 [r8125]
[   15.645234]  rtl8125_powerup_pll+0xa7/0xc0 [r8125]
[   15.645240]  rtl8125_init_one+0x1170/0x2260 [r8125]
[   15.645284]  ? __pfx_rtl8125_init_module+0x10/0x10 [r8125]
[   15.645294]  rtl8125_init_module+0x3d/0xff0 [r8125]
[   15.934097] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[   15.936116] r8125  Copyright (C) 2024 Realtek NIC software team <nicfae@realtek.com> 
[   16.148569] r8125 0000:01:00.0 enp1s0: renamed from eth0
```

I also tried Ubuntu driver in r8125-dkms package (9.011) and is not working.

In all cases the interface is there but it does not detect ethernet connection.

jusii, 2025-04-15:
Hi,

I'll chime in here even though I have different kernel but this sounds exactly the same problem I have. 

But what I found out, that after cold boot (power removed) ethernet works but after soft reboot, it fails. When it fails, I can still see the interface but it wont come up, mac address etc. And when it fails, dmesg is flooded with messages.

Ubuntu 22.04 Asus NUC14MNK-B2
Secure boot disabled

```
uname -a
Linux localhost 6.12.3-061203-generic #202412171635-Ubuntu SMP PREEMPT_DYNAMIC Tue Dec 17 18:32:24 UTC x86_64 x86_64 x86_64 GNU/Linux

lspci -k
01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. Device 8125 (rev 0c)
	Subsystem: ASUSTeK Computer Inc. Device 88e8
	Kernel driver in use: r8125
	Kernel modules: r8169, r8125
	
root@localhost:~# modinfo r8125
filename:       /lib/modules/6.12.3-061203-generic/updates/dkms/r8125.ko
version:        9.015.00-NAPI
license:        GPL
description:    Realtek r8125 Ethernet controller driver
author:         Realtek and the Linux r8125 crew <netdev@vger.kernel.org>
srcversion:     86E9E41D8A4540532B8B7DB
alias:          pci:v000010ECd00003000sv*sd*bc*sc*i*
alias:          pci:v000010ECd00008162sv*sd*bc*sc*i*
alias:          pci:v000010ECd00008125sv*sd*bc*sc*i*
depends:        
name:           r8125
retpoline:      Y
vermagic:       6.12.3-061203-generic SMP preempt mod_unload modversions
```

And these are when it fails:

```
dmesg | grep r8125
[   26.933000] r8125: loading out-of-tree module taints kernel.
[   26.933009] r8125: module verification failed: signature and/or required key missing - tainting kernel
[   26.934138] r8125 Ethernet controller driver 9.015.00-NAPI loaded
[   26.950085] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[   26.950098] r8125  Copyright (C) 2024 Realtek NIC software team <nicfae@realtek.com> 
[   41.740473] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   43.674699] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   44.659483] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   44.660131] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   44.777159] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   44.777284] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   44.790567] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   44.790858] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
..above hundres of lines, and then
[   51.728767] workqueue: rtl8125_esd_task [r8125] hogged CPU for >40000us 4 times, consider switching to WQ_UNBOUND
[   52.002661] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   52.145505] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   52.145655] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   52.201058] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   52.201190] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   52.236272] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   52.236541] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   52.238362] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   52.238713] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   52.290043] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
..above continues until
[   53.837585] workqueue: rtl8125_esd_task [r8125] hogged CPU for >40000us 5 times, consider switching to WQ_UNBOUND
```

And above just repeats

```
dmesg | grep -i realtek
[   26.950098] r8125  Copyright (C) 2024 Realtek NIC software team <nicfae@realtek.com> 
[   36.830563] snd_hda_codec_realtek hdaudioC0D0: autoconfig for ALC256: line_outs=1 (0x21/0x0/0x0/0x0/0x0) type:hp
[   36.830569] snd_hda_codec_realtek hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   36.830571] snd_hda_codec_realtek hdaudioC0D0:    hp_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   36.830573] snd_hda_codec_realtek hdaudioC0D0:    mono: mono_out=0x0
[   36.830574] snd_hda_codec_realtek hdaudioC0D0:    inputs:
[   36.830575] snd_hda_codec_realtek hdaudioC0D0:      Mic=0x19

dmesg | grep r8169
[   26.930385] r8169 0000:01:00.0: enabling device (0000 -> 0003)
[   26.930532] r8169 0000:01:00.0: error -ENODEV: unknown chip XID 688, contact r8169 maintainers (see MAINTAINERS file)
```

And then after cold reboot, when it works:

```
root@localhost:~# dmesg | grep -E 'r8125|r8169'
[   26.756777] r8169 0000:01:00.0: enabling device (0000 -> 0003)
[   26.756933] r8169 0000:01:00.0: error -ENODEV: unknown chip XID 688, contact r8169 maintainers (see MAINTAINERS file)
[   26.759258] r8125: loading out-of-tree module taints kernel.
[   26.759267] r8125: module verification failed: signature and/or required key missing - tainting kernel
[   26.760373] r8125 Ethernet controller driver 9.015.00-NAPI loaded
[   26.776098] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[   26.776137] r8125  Copyright (C) 2024 Realtek NIC software team <nicfae@realtek.com> 
[   40.142228] r8125: eth0: link up

root@localhost:~# dmesg | grep -i realtek
[   26.776137] r8125  Copyright (C) 2024 Realtek NIC software team <nicfae@realtek.com> 
[   36.051876] snd_hda_codec_realtek hdaudioC0D0: autoconfig for ALC256: line_outs=1 (0x21/0x0/0x0/0x0/0x0) type:hp
[   36.051884] snd_hda_codec_realtek hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   36.051887] snd_hda_codec_realtek hdaudioC0D0:    hp_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   36.051889] snd_hda_codec_realtek hdaudioC0D0:    mono: mono_out=0x0
[   36.051890] snd_hda_codec_realtek hdaudioC0D0:    inputs:
[   36.051891] snd_hda_codec_realtek hdaudioC0D0:      Mic=0x19
```

jordicm10, 2025-04-15:
That's very interesting. Thanks a lot for your hint @jusii.

I just shutdown the computer and unplug the power supply a while. After powering up again, the ethernet interface is working. But if I do a soft reboot it fails again.

After all the many unsuccessful test I did with different combinations of kernels and driver versions, I never though that it could be related to the reboot type.

I hope this helps the maintainer to debug the problem.

jordicm10, 2025-04-15:
By the way, I see now some of the same error messages than @jusii 

```
$ sudo dmesg | grep r8125
[    5.908969] r8125: loading out-of-tree module taints kernel.
[    5.956968] r8125 Ethernet controller driver 9.015.00-NAPI loaded
[    5.957072] r8125 0000:01:00.0: enabling device (0000 -> 0003)
[    5.990289] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[    5.990309] r8125  Copyright (C) 2024 Realtek NIC software team <nicfae@realtek.com> 
[    6.044418] r8125 0000:01:00.0 enp1s0: renamed from eth0
[   12.054890] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   12.907543] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   13.908997] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   15.570572] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   15.634195] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   15.634514] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   15.907717] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   16.244419] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   16.513416] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   16.527729] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   16.528604] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   16.907829] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   17.540373] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   17.908111] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   18.392013] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   18.600612] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   18.740807] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   18.772669] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   18.772845] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   18.772958] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   18.908149] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   19.043636] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   19.044020] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   19.578315] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   19.907921] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   20.578946] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   20.600134] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   20.601001] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   20.907983] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   21.560338] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   21.601559] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   21.601894] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   21.604110] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   21.604477] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   21.605501] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   21.605866] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   21.684228] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   21.684581] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   21.857524] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   21.857865] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   21.877122] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   21.877485] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   22.578880] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   23.537154] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   23.847503] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   23.877577] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   23.877927] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   24.578859] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   25.578438] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   25.927912] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   26.578571] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   26.756458] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   26.756796] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   27.534277] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   27.537716] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   27.538594] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   27.738400] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   27.877478] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   27.877806] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[... cut ...]
[   30.636880] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   30.637235] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   30.719207] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   30.719283] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   30.833618] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
[   30.963281] r8125 0000:01:00.0: AER: can't recover (no error_detected callback)
```

kimifoud, 2025-05-18:
I'm not entirely sure if it's the exact same issue, but my Ubuntu 24.04.2 LTS on NUC 14 N150 started losing ethernet connection on reboot after upgrading to `6.8.0-59-generic`, but it was usually fine after `poweroff`. There might be two or even multiple separate issues, but in my case they were somehow related to Active State Power Management.
After hours of experimenting, I may now have found the solution.

`/etc/default/grub`:
```
GRUB_CMDLINE_LINUX_DEFAULT="reboot=acpi pcie_aspm=off"
```
`sudo update-grub`
`sudo reboot`

Now the interface survives reboots.  ¯\_(ツ)_/¯

jordicm10, 2025-05-18:
Thanks @kimifoud !!

I confirm this works.
In my case with processor N150, kernel 6.11.0-25-generic and r8125 driver 9.015.

jusii, 2025-05-19:
No luck here: 6.12.3-061203-generic and r8125 driver 9.015 NUC14MNK-B2 Intel(R) N150

boessu, 2025-05-29:
Hello @jusii 

Maybe you do have similar issues like I have. I had the following error every 2nd reboot (the terminals where getting flooded with this):

```
[  410.073142] enp1s0: pci link is down
(...)
[  412.089213] enp1s0: cmd = 0xff, should be 0x07

```
Try the following in `/etc/default/grub`:
```
GRUB_CMDLINE_LINUX_DEFAULT="r8125.aspm=0 r8125.eee_enable=0 pcie_aspm=off"
```

Then, to store that:
```
sudo update-grub
sudo reboot
```

Everything appears to be working smoothly now. I've gone through several reboots without encountering any issues. It looks like the latest Realtek drivers are having serious trouble with energy-saving features. And it seems as it is not the first time. See:
https://forum.proxmox.com/threads/problem-with-rtl8126-nic-driver-dont-know-how-to-get-it-working.150023/

My machine is an ASUS NUC 14 Essential with N355, which I would like to use as a OMV Server. I'm pretty sure others will come along here because of that problem...

If others do have a comment on that driver issue workaround (!), don't hesitate to answer here.

jusii, 2025-05-30:
@boessu Hey and big thanks! This seems to do it, atleast for 3 reboots and networking still works!

kimifoud, 2025-05-30:
> ```
> GRUB_CMDLINE_LINUX_DEFAULT="r8125.aspm=0 r8125.eee_enable=0 pcie_aspm=off"
> ```

This did the trick!

Kappa71, 2025-06-28:
I had the same problem with a NUC14Essential (N97) and proxmox.
With "GRUB_CMDLINE_LINUX_DEFAULT="r8125.aspm=0 r8125.eee_enable=0 pcie_aspm=off" I confirm that the problem disappear, but the idle power consumption of the nuc rise from 1.9W (minimum that I read on the power meter) to about 4.1W

I downloaded and compiled/installed the latest 9.016.00 version from the realtek site and now it works at every reboot/shutdown even with aspm enabled (and the minium idle power consumption is again 1.9W).
https://www.realtek.com/Download/List?cate_id=584

sullyjman, 2025-10-20:
OS: Ubuntu 24.04.3

Kernel: 6.14.0.33-generic

Motherboard: GIGABYTE B650 GAMING X AX V2 ATX

Network card: r8125

Out of the blue my Ethernet started dropping off (wifi still worked thankfully).  Started searching around and came across this github post.  This is a dual boot system and noticed Windows was having an issue with the network card after sleep too.  

`GRUB_CMDLINE_LINUX_DEFAULT="reboot=acpi pcie_aspm=off"`

The above seems to have cleared up the issue for me.  Rebooted the box multiple times and the network card has stuck around!  Thank you so much!!