awesometic/realtek-r8125-dkms #82: Driver for Kernel 6.14?

Sebiggestboss, 2025-08-06:
Hi!

I ask because Proxmox has just released version 9, which is based on kernel 6.14.

Which driver works with this kernel. The latest version that Realtek has released is version 9.016.00 and it supports the kernel up to 6.12.

timecode, 2025-08-07:
Hi @Sebiggestboss 

The recent release here (https://github.com/awesometic/realtek-r8125-dkms/releases/tag/9.016.00-1) appears to build on Proxmox (v8) kernel `6.14.8-2-bpo12-pve` if that's of help to anyone.

```sh
$ pveversion
pve-manager/8.4.9/649acf70aab54798 (running kernel: 6.14.8-2-bpo12-pve)

$ modinfo r8125
filename:       /lib/modules/6.14.8-2-bpo12-pve/updates/dkms/r8125.ko
version:        9.016.00-NAPI
license:        GPL
description:    Realtek r8125 Ethernet controller driver
author:         Realtek and the Linux r8125 crew <netdev@vger.kernel.org>
srcversion:     F60B45DCE886AB2AF2DC815
alias:          pci:v000010ECd00003000sv*sd*bc*sc*i*
alias:          pci:v000010ECd00008162sv*sd*bc*sc*i*
alias:          pci:v000010ECd00008125sv*sd*bc*sc*i*
depends:
name:           r8125
retpoline:      Y
vermagic:       6.14.8-2-bpo12-pve SMP preempt mod_unload modversions
...
```

I'm about to upgrade to v9, so I'll report back after that 🤞

Sebiggestboss, 2025-08-07:
Hello @timecode,

Thanks for the info! Of course that helps.

I have just received feedback from Realtek support. They also suggested to try version 16. Realtek support responds faster than my local car dealer <3

Would be very nice if you would share your experience! :-)

Sebiggestboss, 2025-08-07:
Hello @timecode 

Thank you very much for the update! 

Will try it on the weekend as well :-)

yakeer, 2025-08-11:
Isn't r8125 should be integrated into r8169 in 6.14?
That at least what I've been reading around the internet.

Sebiggestboss, 2025-08-11:
In the older kernel versions, the 8125 chip was integrated into the r8169 driver. However, the support was not complete. So a trunk with 2 network cards did not work for me.

I have not yet found any indication that the RTL8125BG chipset is better supported by the r8169 driver that comes with  kernel 6.14.

Sebiggestboss, 2025-08-24:
Hi @yakeer  - Thank you for sharing! Good to know.

Sebiggestboss, 2025-09-18:
Did you try to disable IOMMU in the Bios?

Qubitium, 2025-12-31:
> I've had a Cudy switch that my r8125 would constantly crash, once in a couple of weeks, randomally.

Cudy is a glorified linux kernel acting as a soft switch.  My point is a client nic, rtl8125 or anyother nic, crashing your switch has norhing to do with the nic but the bad quality of the switch. We should not conflate nic errors with switch errors. 

> I've switched to a Unifi switch, for 3 months now, not even one crash.

Good switch.  Anything that advertises full line rate on all ports is a proper switch with fixed buffers. Most soft switches like cudy are just routed through the linux kernel. Ok for wifi, not ok for local lan.